PX4/PX4-SITL_gazebo-classic #881: Modeling Engine in Gazebo

tayyabkhalil-313, 2022-07-19:
Hi,
I am trying to add a custom model for SITL. My vehicle doesn’t contain any propellers and so the thrust is produced by the engine. How can i modify the .sdf model file to incorporate the thrust settings of the model?

Jaeyoung-Lim, 2022-07-26:
@tayyabkhalil-313 What exactly do you want to modify in the sdf file? How are you trying to model an engine?

tayyabkhalil-313, 2022-07-26:
Thank you for your response. 
The problem I have is that all the built-in models I have seen in the PX4 have a propeller, and the sdf files are written according to them and using the motor plugin. But my custom plane doesn't contain propellers and I require to model my engine. I want to be able to put the propulsion data of my engine into the sdf file but there's no example of the way to do so.

Jaeyoung-Lim, 2022-07-27:
@tayyabkhalil-313 You cannot model your dynamics using the sdf file. As you can see with the quadrotors, the motor model is modelled as a plugin: 

https://github.com/PX4/PX4-SITL_gazebo/blob/afb2f974f60f9e8f6b08a1f288c42016ec3084f7/models/iris/iris.sdf.jinja#L353-L369

which you can find in: https://github.com/PX4/PX4-SITL_gazebo/blob/main/src/gazebo_motor_model.cpp

If you are not sure how to write your own plugin, there are a lot of tutorials that you can have a look at: https://classic.gazebosim.org/tutorials?tut=plugins_hello_world

Also, if you want to understand more about sdf files: https://classic.gazebosim.org/tutorials?tut=build_model

Jaeyoung-Lim, 2022-08-02:
Closing as the issuer haven't responded

@tayyabkhalil-313 Feel free to reopen the issue

closed 2022-08-02 by Jaeyoung-Lim