PX4/PX4-SITL_gazebo-classic #883: multi_uav_mavros_sitl.launch file Unable to launch iris_fpv_cam or iris_depth_camera

enginksz, 2022-07-22:
Hello,

When I use the iris_fpv_cam model in the field specified as the vehicle model in the multi_uav_mavros_sitl.launch file, I cannot get a connection via QGround.

Since it also runs the mavros plugin, no message is received from any topic via mavros.

There is no problem when we use "iris" in the multi_uav_mavros_sitl.launch file, but when we change the model, the connection over QGS does not appear.

How we can solve this problem ?

**multi_uav_mavros_sitl.launch** 

<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="**iris_fpv_cam**"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/**single_vehicle_spawn**.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="14530"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    
    <group ns="uav1">
        
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14561"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="14531"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
</launch>

[image]
[image]

enginksz, 2022-07-24:
Hello again,

I've seen this issue in other issues and fixed it.

Thanks

closed 2022-07-24 by enginksz

Jaeyoung-Lim, 2022-07-24:
@enginksz Could you please link the solution to the thread?

enginksz, 2022-07-24:
Hi, 
I followed the [link](https://discuss.px4.io/t/how-to-add-a-ros-camera-to-iris-for-gazebo-simulation/5118)  to add a camera at the bottom of Xacro file of the iris itself.

Jaeyoung-Lim, 2022-07-24:
@enginksz Thanks!