PX4/PX4-SITL_gazebo-classic #889: Can we change mavlink tcp port to which gazebo connects ?

airsimdevdd, 2022-08-01:
I want gazebo to connect to different tcp port other than tcp  4560 ,px4 must open 4560 I want to intercept the msgs and redirect it to gazebo

Jaeyoung-Lim, 2022-08-02:
@airsimdevdd Yes, you can change the port by configuring the tcp port of the plugin: https://github.com/PX4/PX4-SITL_gazebo/blob/f71680afab60da2d72bcbddbed0b289cd0cd7814/models/iris/iris.sdf.jinja#L453

airsimdevdd, 2022-08-03:
> @airsimdevdd Yes, you can change the port by configuring the tcp port of the plugin:
> 
> https://github.com/PX4/PX4-SITL_gazebo/blob/f71680afab60da2d72bcbddbed0b289cd0cd7814/models/iris/iris.sdf.jinja#L453

I am launching Gazebo with ros wrappers, I have already changed TCP port of the plugin yet everything remains the same.

Jaeyoung-Lim, 2022-08-03:
@airsimdevdd Have you checked if the sdf file generated have the tcp port changes you applied? If not you need to figure out what your ros wrapper is doing with the sdf generation

airsimdevdd, 2022-08-03:
> @airsimdevdd Have you checked if the sdf file generated have the tcp port changes you applied? If not you need to figure out what your ros wrapper is doing with the sdf generation

Yes I have checked, sdf file does have the same tcp port. Any more suggestions...??

Jaeyoung-Lim, 2022-08-03:
@airsimdevdd Then it means your change did not get applied? It is not possible that nothing changes if the tcp port has changed in the sdf file and the simulation is using it. The plugin configures the port

Jaeyoung-Lim, 2022-08-24:
Closing as there was no response

closed 2022-08-24 by Jaeyoung-Lim