PX4/PX4-SITL_gazebo-classic #906: SDF to URDF convertion problem - EKF switching, no GPS

WingRS, 2022-08-30:
Hi,
I've tried to convert the sdf to urdf for attaching some of the gazebo plugins to the model and have faced a couple of problems.
The main is that the px4 refuses to take off due to EKF switching from GPS to baro and then back and an issue with the accelerometer.
The model is here [https://pastebin.com/fTPXN6bA](https://pastebin.com/fTPXN6bA).
The log of pixhawk during the start 
```
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [commander] Mission #2 loaded, 15 WPs, curr: 13
INFO  [init] Mixer: etc/mixers-sitl/plane_sitl.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [logger] logger started (mode=all)
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-08-30/10_20_18.ulg
INFO  [logger] Opened full log file: ./log/2022-08-30/10_20_18.ulg
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] 392000: GPS checks passed
INFO  [ecl/EKF] 3256000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 3256000: reset position to last known position
INFO  [ecl/EKF] 3256000: reset velocity to zero
INFO  [ecl/EKF] 4924000: reset position to GPS
INFO  [ecl/EKF] 4924000: reset velocity to GPS
INFO  [ecl/EKF] 4924000: starting GPS fusion
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
pxh> WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps

pxh> commander check
WARN  [PreFlightCheck] Preflight Fail: Accel Range, hold still on arming
WARN  [PreFlightCheck] Primary accelerometer not found
INFO  [commander] Preflight check: FAILED
INFO  [commander] Prearm check: OK
INFO  [HealthFlags] DEVICE              STATUS
INFO  [HealthFlags] ----------------------------------
INFO  [HealthFlags] GYRO:               EN      OK
INFO  [HealthFlags] ACC:                EN      OFF
INFO  [HealthFlags] MAG:                EN      OK
INFO  [HealthFlags] PRESS:              EN      OK
INFO  [HealthFlags] AIRSP:              EN      OK
INFO  [HealthFlags] GPS:                EN      OK
INFO  [HealthFlags] OPT:
INFO  [HealthFlags] VIO:
INFO  [HealthFlags] LASER:
INFO  [HealthFlags] GTRUTH:
INFO  [HealthFlags] RATES:
INFO  [HealthFlags] ATT:
INFO  [HealthFlags] YAW:
INFO  [HealthFlags] ALTCTL:
INFO  [HealthFlags] POS:
INFO  [HealthFlags] MOT:
INFO  [HealthFlags] RC  :               EN      OFF
INFO  [HealthFlags] GYRO2:
INFO  [HealthFlags] ACC2:
INFO  [HealthFlags] MAG2:
INFO  [HealthFlags] GEOFENCE:
INFO  [HealthFlags] AHRS:               EN      ERR
INFO  [HealthFlags] TERRAIN:
INFO  [HealthFlags] REVMOT:
INFO  [HealthFlags] LOGGIN:
INFO  [HealthFlags] BATT:
INFO  [HealthFlags] PROX:
INFO  [HealthFlags] SATCOM:
INFO  [HealthFlags] PREARM:             EN      ERR
INFO  [HealthFlags] OBSAVD:
pxh> WARN  [ecl/EKF] baro hgt timeout - reset to gps

pxh> commander takeoff
pxh> WARN  [PreFlightCheck] Preflight Fail: Accel Range, hold still on arming
WARN  [PreFlightCheck] Primary accelerometer not found
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to gps
 
```
I've been trying to change the model, but so far no luck.
The gazebo gps plugin seems to be working and publishing the data ('ve printed some logs), but `/mavros/global_position/global` is empty.

Jaeyoung-Lim, 2022-08-30:
@WingRS Could you explain why would you want to convert sdf files to urdf? 

SDFs are something derived from urdfs, that provide more advanced syntaxes and functionalities

WingRS, 2022-09-08:
The thing is, I am using a macro for some other projects, and we have developed a set of macros. I haven't had a big issue with the SDF to pdf conversion, but this one is quite annoying; in SDF, it seems to work perfectly.

Jaeyoung-Lim, 2022-09-11:
@WingRS It is because certain syntaxes of sdf cannot translate directly to a urdf and hence a xacro.

For this reason, we have switched to using jinja templates rather than xacros for sdf generation