PX4/PX4-SITL_gazebo-classic #912: cant compile in ubuntu 20.04

jorgeneo560, 2022-09-23:
i cant compile, theres a lot of problems in this ubuntu version, i was able to fixed the most part but now i am stuck here
<pre>-- install-prefix: /usr
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- Found TinyXML: /usr/lib/x86_64-linux-gnu/libtinyxml.so  
-- Checking for module &apos;gstreamer-1.0 &gt;= 1.0&apos;
--   No package &apos;gstreamer-1.0&apos; found
-- Checking for module &apos;gstreamer-base-1.0 &gt;= 1.0&apos;
--   No package &apos;gstreamer-base-1.0&apos; found
-- Checking for module &apos;gstreamer-app-1.0 &gt;= 1.0&apos;
--   No package &apos;gstreamer-app-1.0&apos; found
-- Found GStreamer: GSTREAMER_INCLUDE_DIRS;GSTREAMER_LIBRARIES;GSTREAMER_VERSION;GSTREAMER_BASE_INCLUDE_DIRS;GSTREAMER_BASE_LIBRARIES (Required is at least version &quot;1.0&quot;) 
-- Found unsuitable Qt version &quot;&quot; from NOTFOUND
Qt QTGUI library not found.
Qt QTCORE library not found.
-- Checking for module &apos;OGRE&apos;
--   No package &apos;OGRE&apos; found
<font color="#C01C28">CMake Error at CMakeLists.txt:103 (add_subdirectory):</font>
<font color="#C01C28">  The source directory</font>

<font color="#C01C28">    /home/x/PX4/PX4-SITL_gazebo/external/OpticalFlow</font>

<font color="#C01C28">  does not contain a CMakeLists.txt file.</font>

-- Could NOT find MAVLink (missing: MAVLINK_INCLUDE_DIRS) (found version &quot;2.0&quot;)
-- catkin DISABLED
<font color="#C01C28">CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):</font>
<font color="#C01C28">  Could NOT find Protobuf (missing: Protobuf_LIBRARIES Protobuf_INCLUDE_DIR)</font>
<font color="#C01C28">Call Stack (most recent call first):</font>
<font color="#C01C28">  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)</font>
<font color="#C01C28">  /usr/share/cmake-3.22/Modules/FindProtobuf.cmake:650 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)</font>
<font color="#C01C28">  CMakeLists.txt:145 (find_package)</font>

-- Configuring incomplete, errors occurred!
See also &quot;/home/x/PX4/PX4-SITL_gazebo/build/CMakeFiles/CMakeOutput.log&quot;.
</pre>
dont know what to do, i need to setup a simulation to test flight paths generated with qground

tayyabkhalil-313, 2022-10-03:
Did you install the dependencies for running the simulation described [here](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets)? Did running the script _ubuntu.sh_ throw up any errors?

Jaeyoung-Lim, 2022-10-03:
@jorgeneo560 You need to update your submodule

Jaeyoung-Lim, 2022-10-06:
@Katawann I don't think your issue is related.
The builds are checked in the CI, so I suspect there is something wrong specific to your setup.

Why not use the setup script (`Tools/setup/ubuntu.sh`) to install the dependencies?

closed 2022-12-20 by Jaeyoung-Lim