PX4/PX4-SITL_gazebo-classic #920: Building PX4 Software

HadesGuy, 2022-10-26:
Following the tutorials on https://docs.px4.io/main/en/dev_setup/building_px4.html I am using Ubuntu 18.04 VM and trying to do a basic install and when i put the command `make px4_fmu-v5_default` am getting the error bellow. 
Does anyone knows how to fix this or any workarounds on it?
Thank you  

sim@sim-VirtualBox:~/PX4-Autopilot$ make px4_fmu-v5_default
-- PX4 version: v1.13.0-beta1-1062-g740d2fccb1
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- PX4 config file: /home/sim/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 6.3.1
-- The C compiler identification is GNU 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- drivers/px4io: ROMFS including px4_io-v2_default
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Using C++11
-- Release build type: MinSizeRel
CMake Warning at src/modules/microdds_client/CMakeLists.txt:35 (message):
  skipping microdds_client, Micro-XRCE-DDS-Client needs to be fixed to work
  with CMAKE_VERSION 3.10.2

-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32f7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_fmu-v5_bootloader.bin -> /etc/extras/px4_fmu-v5_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sim/PX4-Autopilot/build/px4_fmu-v5_default
[0/1175] git submodule src/drivers/gps/devices
[8/1175] git submodule platforms/nuttx/NuttX/nuttx
[12/1175] git submodule src/drivers/uavcan/libuavcan
[13/1175] git submodule src/modules/mavlink/mavlink
[14/1175] Generating Mavlink standard: ...k/message_definitions/v1.0/standard.xml
Traceback (most recent call last):
  File "/home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
FAILED: mavlink/standard/standard.h 
cd /home/sim/PX4-Autopilot/build/px4_fmu-v5_default/src/modules/mavlink && /usr/bin/python3 /home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py --lang C --wire-protocol 2.0 --output /home/sim/PX4-Autopilot/build/px4_fmu-v5_default/mavlink /home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
[20/1175] Generating parameters.xml
ninja: build stopped: subcommand failed.
Makefile:227: recipe for target 'px4_fmu-v5_default' failed
make: *** [px4_fmu-v5_default] Error 1
sim@sim-VirtualBox:~/PX4-Autopilot$

tayyabkhalil-313, 2022-11-08:
There seems to be a issue with dependencies as all the modules are not installed. 

> ModuleNotFoundError: No module named 'future'

To resolve this, run `pip install future`

HadesGuy, 2022-11-18:
Hi 
I ran the command and got the error below:

sim@ubuntu:~/PX4-Autopilot$ sudo su
[sudo] password for sim: 
root@ubuntu:/home/sim/PX4-Autopilot# make px4_fmu-v5_default
[0/1] Re-running CMake...
-- PX4 version: v1.13.0-beta1-1080-gba3f3935ab
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/sim/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

-- Configuring incomplete, errors occurred!
See also "/home/sim/PX4-Autopilot/build/px4_fmu-v5_default/CMakeFiles/CMakeOutput.log".
See also "/home/sim/PX4-Autopilot/build/px4_fmu-v5_default/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
/usr/bin/cmake -S/home/sim/PX4-Autopilot -B/home/sim/PX4-Autopilot/build/px4_fmu-v5_default
ninja: error: rebuilding 'build.ninja': subcommand failed
make: *** [Makefile:227: px4_fmu-v5_default] Error 1
root@ubuntu:/home/sim/PX4-Autopilot# pip install future
Collecting future
  Downloading future-0.18.2.tar.gz (829 kB)
     |████████████████████████████████| 829 kB 5.9 MB/s 
Building wheels for collected packages: future
  Building wheel for future (setup.py) ... done
  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491058 sha256=05a9a8ec0a6024113cf90675468a07c0ad16e7170886f158e14e7b1c6bee28cb
  Stored in directory: /root/.cache/pip/wheels/8e/70/28/3d6ccd6e315f65f245da085482a2e1c7d14b90b30f239e2cf4
Successfully built future
Installing collected packages: future
Successfully installed future-0.18.2
root@ubuntu:/home/sim/PX4-Autopilot# make px4_fmu-v5_default
[0/1] Re-running CMake...
-- PX4 version: v1.13.0-beta1-1080-gba3f3935ab
-- PX4 config file: /home/sim/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

-- Configuring incomplete, errors occurred!
See also "/home/sim/PX4-Autopilot/build/px4_fmu-v5_default/CMakeFiles/CMakeOutput.log".
See also "/home/sim/PX4-Autopilot/build/px4_fmu-v5_default/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
/usr/bin/cmake -S/home/sim/PX4-Autopilot -B/home/sim/PX4-Autopilot/build/px4_fmu-v5_default
ninja: error: rebuilding 'build.ninja': subcommand failed
make: *** [Makefile:227: px4_fmu-v5_default] Error 1
root@ubuntu:/home/sim/PX4-Autopilot#

tayyabkhalil-313, 2022-11-18:
Yes it does. The firmware has been successfully built.