PX4/PX4-SITL_gazebo-classic #927: PX4 SITL Gazebo some plugins not working

saksham2001, 2022-11-09:
I am trying to use the PX4 SITL simulation with gazebo. I wanted to try rgb camera and depth camera attached to the drone and process that using ROS 1. I saw models for `iris_fpv_cam` and `iris_depth_camera` models already present in `PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo/models`. I was not directly able to launch these using the command `make px4_sitl gazebo_iris_fpv_cam`, so did the following:
- created a airframe in `ROMFS/px4fmu_common/init.d-posix` as `5001_iris_depth_camera` and used the same files as in the iris_rplidar frame.
- added this airframe in `ROMFS/px4fmu_common/init.d-posix/CmakeLists.txt`

Now after launching using the command `make px4_sitl gazebo_iris_depth_camera`, everything works. Gazebo launches and I can see the depth camera on the iris frame. But there are no rostopics published for the depth camera. Same thing for `iris_fpv_cam` and even `iris_rplidar` (which is already present and does not require any changes with ROMFS). I have tried launching roscore, or the mavros launch file before starting this and it doesn't help. Ther

Operating System: Ubuntu 22.04

ROS: Noetic

Results after `gazebo topics -l`
```
/gazebo/default/atmosphere
/gazebo/default/default/depth_camera/cmd
/gazebo/default/diagnostics
/gazebo/default/factory
/gazebo/default/factory/light
/gazebo/default/gui
/gazebo/default/gzclient_camera/cmd
/gazebo/default/iris_depth_camera/baro
/gazebo/default/iris_depth_camera/camera/link/irlock
/gazebo/default/iris_depth_camera/depth_camera/link/depth_camera/image
/gazebo/default/iris_depth_camera/depth_camera/link/wrench
/gazebo/default/iris_depth_camera/gazebo/command/motor_speed
/gazebo/default/iris_depth_camera/groundtruth
/gazebo/default/iris_depth_camera/iris/base_link/wrench
/gazebo/default/iris_depth_camera/iris/gps0/link/gps
/gazebo/default/iris_depth_camera/iris/gps0/link/wrench
/gazebo/default/iris_depth_camera/iris/imu/link/imu/imu
/gazebo/default/iris_depth_camera/iris/imu/link/wrench
/gazebo/default/iris_depth_camera/iris/mag/link/magnetometer
/gazebo/default/iris_depth_camera/iris/mag/link/wrench
/gazebo/default/iris_depth_camera/iris/rotor_0/wrench
/gazebo/default/iris_depth_camera/iris/rotor_1/wrench
/gazebo/default/iris_depth_camera/iris/rotor_2/wrench
/gazebo/default/iris_depth_camera/iris/rotor_3/wrench
/gazebo/default/iris_depth_camera/link/gps0
/gazebo/default/iris_depth_camera/link/imu
/gazebo/default/iris_depth_camera/link/mag
/gazebo/default/iris_depth_camera/px4flow/link/opticalFlow
/gazebo/default/iris_depth_camera/vision_odom
/gazebo/default/iris_depth_camera::iris::imu::link
/gazebo/default/joint
/gazebo/default/light/modify
/gazebo/default/log/control
/gazebo/default/log/status
/gazebo/default/model/info
/gazebo/default/model/modify
/gazebo/default/physics
/gazebo/default/physics/contacts
/gazebo/default/playback_control
/gazebo/default/pose/info
/gazebo/default/pose/local/info
/gazebo/default/pose/modify
/gazebo/default/request
/gazebo/default/response
/gazebo/default/roads
/gazebo/default/scene
/gazebo/default/selection
/gazebo/default/sensor
/gazebo/default/skeleton_pose/info
/gazebo/default/sky
/gazebo/default/undo_redo
/gazebo/default/user_camera/joy_pose
/gazebo/default/user_camera/joy_twist
/gazebo/default/user_camera/pose
/gazebo/default/user_cmd
/gazebo/default/user_cmd_stats
/gazebo/default/visual
/gazebo/default/wind
[... 210 lines cut ...]
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint_desired 0
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-11-09/18_55_08.ulg	
INFO  [logger] Opened full log file: ./log/2022-11-09/18_55_08.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [tone_alarm] home set
INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [commander] Ready for takeoff!
INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [mavlink] partner IP: 127.0.0.1
ERROR [timesync] Time jump detected. Resetting time synchroniser.
ERROR [timesync] Time jump detected. Resetting time synchroniser.
```

Looking forward to some help. Please let me know if any additional information is required.

MichaelBosello, 2023-01-10:
Hi, I have the same problem, did you find a solution?

sibujacob, 2023-04-21:
i also have the same issue....do you have a solution

reo7749, 2023-09-20:
i have the same issue, ubuntu 20.04

AnishMathewOommen, 2024-04-07:
I have the same issue in Ubuntu 20.04

karzito, 2024-10-29:
i hvae the same issue

Prit44421, 2024-12-22:
i have the same issue