PX4/PX4-SITL_gazebo-classic #938: I can't get the orientation of gimbal (Typhoon H480 in Gazebo)

boris-gu, 2022-12-22:
I may be wrong in my conclusions because I haven't found any documentation.

I am using ROS Noetic and running a Gazebo with a Typhoon H480 model. This model uses a plugin "libgazebo_gimbal_controller_plugin.so "  
I need to get the angles of gimbal. I found three topics that seem to be useful to me:  
```
/mavros/mount_control/command
/mavros/mount_control/orientation
/mavros/mount_control/status
```
But when I listen to topics `/mavros/mount_control/orientation` and `/mavros/mount_control/status`, there are no messages there, instead I get:
```
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

What am I doing wrong? How do I get the gimbal angles?

Jaeyoung-Lim, 2022-12-22:
@boris-gu Are you sure your mavros instance is connected to the correct udp port that has gimbal messages?

boris-gu, 2022-12-22:
@Jaeyoung-Lim I do not know how to check it, but I think that everything is connected correctly because:  
1. I can get data, for example, from the topic `/mavros/local_position/pose`  
2. Typhoon H480 is a standard PX4 model, I haven't changed anything in it

Jaeyoung-Lim, 2022-12-22:
@boris-gu gimbal topics are normally published over a different port, and I am not sure if the expectation that the mavros mount control plugin works out of the box with the typhoon model is correct

boris-gu, 2022-12-22:
@Jaeyoung-Lim Okay, thanks for the answer.  
I have a few more questions left:
1. Do I understand correctly that it is impossible to configure this in the simulator / it will require a lot of work?
2. What should I do to make it work on a real drone?
3. I could not find documentation on these topics and how to get data from gimbal via MAVROS. Where can I read about it?

Jaeyoung-Lim, 2022-12-22:
> 1. Do I understand correctly that it is impossible to configure this in the simulator / it will require a lot of work?

Not it should work. I was just pointing out that the fact that you get imu data and no mount control probably means that your gimbal control mavlink messages are not reaching the flight controller. Therefore I would check that.

I don't think this would require any work, but just making sure the configurations are correct.

> What should I do to make it work on a real drone?

This highly depends on which gimbal interface you are using. You can read more on https://docs.px4.io/main/en/advanced/gimbal_control.html

Here is an example that I wrote before, but not sure if it works now with the latest stable release (not tested): https://github.com/Jaeyoung-Lim/mavros_humantracking

You need to understand how the gimbal messaging works with mavlink. And then it would become clearer why it does or does not work.

closed 2022-12-22 by boris-gu