PX4/PX4-SITL_gazebo-classic #939: Namespace to camera plugin

Cavalletta98, 2022-12-27:
Hi, i0m using the PX4 avoidance software and i noticed that the camera topic is without namespace. How i can add a namespace such date i can simulate multiple cameras?

Jaeyoung-Lim, 2022-12-27:
@Cavalletta98 Which namespace are you talking about?

Cavalletta98, 2022-12-27:
The problem is related to this plugin:
`<plugin filename="libgazebo_ros_openni_kinect.so" name="camera_controller">
          <cameraName>camera</cameraName>
          <alwaysOn>true</alwaysOn>
          <updateRate>10</updateRate>
          <pointCloudCutoff>0.2</pointCloudCutoff>
          <pointCloudCutoffMax>65.535</pointCloudCutoffMax>
          <imageTopicName>rgb/image_raw</imageTopicName>
          <cameraInfoTopicName>rgb/camera_info</cameraInfoTopicName>
          <depthImageTopicName>depth/image_raw</depthImageTopicName>
          <depthImageCameraInfoTopicName>depth/camera_info</depthImageCameraInfoTopicName>
          <pointCloudTopicName>depth/points</pointCloudTopicName>
          <frameName>camera_link</frameName>
          <distortion_k1>0.0</distortion_k1>
          <distortion_k2>0.0</distortion_k2>
          <distortion_k3>0.0</distortion_k3>
          <distortion_t1>0.0</distortion_t1>
          <distortion_t2>0.0</distortion_t2>
        </plugin>`
If i use the same plugin, launched multiple times, i cannot get topics with different names (one for each vehicle). How i can do it?

Jaeyoung-Lim, 2022-12-27:
You need to set different topic names for different vehicles

Cavalletta98, 2022-12-27:
[goap_avoid.txt](https://github.com/PX4/PX4-SITL_gazebo/files/10308136/goap_avoid.txt)
@Jaeyoung-Lim This is my launch file. Is it correct like this?

Jaeyoung-Lim, 2022-12-27:
@Cavalletta98 It is hard to comment whether the launchfile is correct by just looking at the file.

As long as you understood how the topic names should be configured I guess it should work

Cavalletta98, 2022-12-27:
@Jaeyoung-Lim This launch file Is not working. I'm gettin the camera topico without namespace. How can i solve It?