PX4/PX4-SITL_gazebo-classic #940: Multi-vehicle Gimbal Control Failed

andy-zhuo-02, 2022-12-29:
Hello,
I'm trying to control control gimbals for different uavs(typhoon_h480_0, typhoon_1 ... etc.) at the same time. There is something wrong when I use topic `/typhoon_h480_0/mavros/mount_control/command` to control the gimbal of  typhoon_h480_0. In Gazebo simulator, the moving gimbal is uav_1(in later attempts, it also occurs to uav2), instead of desired uav_0 (in later attempts, it also occurs to uav2).

[image]

And rosgraph shows the node and topic.

[image]

PX4 current version is 1.13.2, Gazebo version is  11 and ROS is melodic version.
The launch file I used is based on origin launch `multi_uav_siti_sdf.launch` and I just change the vehicle name from plane to typhoon_h480, along with group ns.

is there something I missed? Any suggestion would help,  thx.

By the way, I have searched issues and found that issue #540 (2 years ago) situation is quite similar to me. However, the bug seems to have been fixed.

Jaeyoung-Lim, 2023-02-20:
@andy-zhuo-02 The launchfile in https://github.com/Jaeyoung-Lim/mavros_humantracking might be helpful for you

shupx, 2023-07-17:
@andy-zhuo-02 The `gazebo_gimbal_controller_plugin` in PX4 v1.13 used a fixed udp port ```dest_addr.sin_port = htons(13030)```, which is not applicable to multiple gimbals situation:
https://github.com/PX4/PX4-SITL_gazebo-classic/blob/48440d7b5c78a21182415266334981f1163f4b2c/src/gazebo_gimbal_controller_plugin.cpp#L743-L757

This bug has been fixed in PX4 v1.14 or higher version, which adds a sdf element 'udp_gimbal_port_remote' to determine the gimbal control port:
https://github.com/PX4/PX4-SITL_gazebo-classic/blob/1c5818e56c2f54ca16f465fbf43580b1b6f3ece0/src/gazebo_gimbal_controller_plugin.cpp#L361-L367 https://github.com/PX4/PX4-SITL_gazebo-classic/blob/1c5818e56c2f54ca16f465fbf43580b1b6f3ece0/src/gazebo_gimbal_controller_plugin.cpp#L768

Therefore, use the `gazebo_gimbal_controller_plugin` of PX4 v1.14 as listed above, and add a `udp_gimbal_port_remote` element to the gimbal plugin in typhoon_h480 sdf model:
https://github.com/PX4/PX4-SITL_gazebo-classic/blob/1c5818e56c2f54ca16f465fbf43580b1b6f3ece0/models/typhoon_h480/typhoon_h480.sdf.jinja#L1433-L1439, which should be like:
```xml
 <plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'> 
   <udp_gimbal_port_remote> 13030 </udp_gimbal_port_remote>
   <joint_yaw>typhoon_h480::cgo3_vertical_arm_joint</joint_yaw> 
   <joint_roll>typhoon_h480::cgo3_horizontal_arm_joint</joint_roll> 
   <joint_pitch>typhoon_h480::cgo3_camera_joint</joint_pitch> 
```
Change the `udp_gimbal_port_remote` for multiple typhoon models as `13030`, `13031`, etc., which means you should create multiple typhoon_h480 sdf model and launch them seperately.

Then, you can control multiple gimbals simultaneously.

andy-zhuo-02, 2023-07-25:
> 

Your method solved my problem perfectly. Thank you.

closed 2023-07-25 by andy-zhuo-02