PX4/PX4-SITL_gazebo-classic #943: Video streaming doesn't work on Mac OS with gazebo_gst_camera_plugin.cpp

murphym18, 2023-01-05:
On MacOS (Arm) the simulated video stream doesn't work.

To re-create the problem:

1. create a PX4 dev environment on Mac OS
2. run `make px4_sitl gazebo_typhoon_h480`

The simulation will start but the video stream doesn't work.

After doing some digging. We found that `gst_element_factory_make` is returning `null` for all of these:

- `source`
- `converter`
- `encoder`
- `payloader`
- `sink`

Here's a link to the code:

https://github.com/PX4/PX4-SITL_gazebo/blob/05348f9c22f31fe214732897411c279aa51e8dde/src/gazebo_gst_camera_plugin.cpp#L65-L94

Jaeyoung-Lim, 2023-01-06:
@murphym18 Could you maybe check whether the gstreamer plugins like `x264enc` works on your system?

https://gstreamer.freedesktop.org/documentation/x264/index.html?gi-language=c

JasonBrauer, 2023-01-06:
@Jaeyoung-Lim I've replicated the same issue @murphym18 describes above. I've also built gstreamer pipelines on the same Arm Mac that successfully utilize the `x264enc` and `videoconvert` plugins.   

Could there be an issue with referencing the gstreamer libraries during the build process kicked off by `make`?