PX4/PX4-SITL_gazebo-classic #951: unable to read barometer sensor data

airsimdevdd, 2023-01-28:
I have a gazebo in which i have many sensors  like imu ,gps,magnetometer and their data is published in gazebo::msgs but barometer data is not publishing in gazebo::msgs it is publishing in sensor_msgs::msgs but unable to read the data any way to get barometer and other sensor data get  published 

Thanks in Advance

Jaeyoung-Lim, 2023-02-20:
@airsimdevdd You can subscribe to the gazebo messages.

If you are not sure how to do this, here is some tutorial that you can follow: https://classic.gazebosim.org/tutorials?tut=topics_subscribed&cat=transport

airsimdevdd, 2023-02-21:
@Jaeyoung-Lim barometer is being published in sensor_msgs::msgs im unable to read it getting error like unable to prase the data

airsimdevdd, 2023-02-21:
unable to parse message of type [sensors_msgs.msgs.Pressure]

Jaeyoung-Lim, 2023-02-21:
@airsimdevdd
It is really hard to provide any help with the limited information you are providing, and it discourages me to provide help

Please provide more context.( Where is that error occuring, how are you trying to parse the message, why are you trying yo parse the message, etc)

Jaeyoung-Lim, 2023-03-15:
@airsimdevdd You need to modify the plugin to publish the data to gazebo topics if you want to do that. The reason we don't use the standard message is that the standard message lacks a lot of essential data to simulate the barometer(e.g. temperature)