PX4/PX4-SITL_gazebo-classic #952: what is exact behind the gazebo motor model plugin

airsimdevdd, 2023-01-28:
what is the exact formulas for calculating the parameters in gazebo model plugins and what are their units i need detailed description 

thanks in advance

Jaeyoung-Lim, 2023-01-28:
https://github.com/PX4/PX4-SITL_gazebo-classic/blob/9343aaf4e275db48fce02dd25c5bd8273c2d583a/src/gazebo_motor_model.cpp#L196-217

airsimdevdd, 2023-01-28:
@Jaeyoung-Lim  how to calculate  angular  velocity motor constant and others ,what are units for prop dia

Jaeyoung-Lim, 2023-01-28:
@airsimdevdd The resulting force is in newtons, so you can check what the units for the factors are

airsimdevdd, 2023-01-28:
I need some clarity  on formulas  and propeller diameter in rotordrag coefficient can u post it

Jaeyoung-Lim, 2023-01-28:
@airsimdevdd Unfortunately there is no material available with the exact formulas, therefore I cannot post it

airsimdevdd, 2024-11-04:
@Jaeyoung-Lim  any update  on  this ,  when importing the nano drones like 1 kg drone to Gazebo  I'm getting a problem with formulas from this https://github.com/PX4/PX4-SITL_gazebo-classic/issues/110

Jaeyoung-Lim, 2024-11-04:
@airsimdevdd Have you looked at the code on how the thrust is computed?

airsimdevdd, 2024-11-07:
force=rot_velocity 2 ×motor_constant i found this in code

airsimdevdd, 2024-11-07:
> @airsimdevdd Have you looked at the code on how the thrust is computed?

force=rot_velocity 2 ×motor_constant this is what i found