PX4/PX4-SITL_gazebo-classic #954: Unable to run make px4_sitl gazebo

JoshikaNetha, 2023-02-08:
Hi,

I'm getting below error while running make px4_sitl gazebo
[0/2] Re-checking globbed directories...
ninja: error: unknown target 'gazebo', did you mean 'geo'?
make: *** [Makefile:232: px4_sitl] Error 1

please help me with this issue.

Thanks in advance.

hamishwillee, 2023-02-08:
I'm sure the experts will answer this, but to help them you should outline what OS, OS version and version of PX4 you are running.

JoshikaNetha, 2023-02-08:
Sure , I'll provide  OS version is ubuntu-20.04 and PX4 is1.14.0

Deltarr350, 2023-02-10:
Hi, I have the exact same problem here, running on Ubuntu 18.04. Thank you in advance.

hamishwillee, 2023-02-10:
Hmmm. That should work. Can you try `make px4_sitl gazebo-classic`.

Deltarr350, 2023-02-10:
Thanks for the quick response, it still doesn't work though. The error message is as follows:

ninja: error: unknown target 'gazebo-classic'
Makefile:232: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

hamishwillee, 2023-02-10:
@Jaeyoung-Lim Perhaps one for you

JoshikaNetha, 2023-02-10:
After git pull origin,  and updating submodule, I got gazebo_classic and its working.

closed 2023-02-10 by hamishwillee

rosiakpiotr, 2023-03-19:
Sorry, but the issue seems to be still here, after cloning fresh copy of repository and running Tools/setup/ubuntu.sh on Ubuntu 22 I get the same error

```
➜  PX4-Autopilot git:(main) ✗ make px4_sitl gazebo-classic
ninja: error: unknown target 'gazebo-classic'
make: *** [Makefile:232: px4_sitl] Error 1
```

Jaeyoung-Lim, 2023-03-19:
@rosiakpiotr 

```
make px4_sitl gazebo-classic_iris
```

rosiakpiotr, 2023-03-19:
Still no luck. I had the same problem on Docker container based on `px4io/px4-dev-simulation-focal` on my desktop. I was sure something was off with the image so I tried installing it directly on WSL2 (forgot to mention I work on WSL2 + Docker, no native Linux), it worked on my Win11 desktop so I went ahead and repeated the same steps on WSL2 on my Win11 laptop, but it keeps failing here.

Jaeyoung-Lim, 2023-03-19:
@rosiakpiotr You need to check your PX4 version

rosiakpiotr, 2023-03-19:
My findings indicate that it builds just fine on Ubuntu 20 LTS but fails on Ubuntu 22 LTS. I have `main` branch selected on PX4 repo, switching to `v1.13.3` doesn't help either. For my case that's enough - I will use Ubuntu20, but let's stay aware of the problem on U22.

Jaeyoung-Lim, 2023-03-19:
@rosiakpiotr That is because on 22.04, you don't have gazebo-classic available

rosiakpiotr, 2023-03-19:
You are right. Sorry for the confusion I should have noticed that it is not advised to build for Gazebo Classic on 22.04. My apologies.

Jaeyoung-Lim, 2023-03-19:
@rosiakpiotr No worries

arielc-brillianetor, 2023-07-05:
Hi, I'm getting the same problem (running on Ubuntu 20.04):
`~/PX4-Autopilot$ make px4_sitl gazebo-classic
-- PX4 version: v1.14.0-beta2-305-ga6502c9f2b (1.14.0)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/ariel/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/cc
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/lib/ccache/cc
-- Check for working C compiler: /usr/lib/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- ccache enabled via symlink (/usr/lib/ccache/c++ -> /usr/bin/ccache)
-- Looking for gz-transport12 -- found version 12.2.0
-- Searching for dependencies of gz-transport12
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Found UUID: TRUE  
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.0.0
-- Searching for dependencies of gz-utils2-cli
-- Looking for gz-msgs9 -- found version 9.4.0
-- Searching for dependencies of gz-msgs9
-- Looking for gz-math7 -- found version 7.2.0
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.0.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Found Java: /usr/bin/java (found version "17.0.7") 
-- ROMFS: ROMFS/px4fmu_common
Architecture:  amd64
==> CPACK_INSTALL_PREFIX = @DEB_INSTALL_PREFIX@
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ariel/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'gazebo-classic'
make: *** [Makefile:232: px4_sitl] Error 1
`

I tried adding the iris as well after the classic but no help.
Any ideas?

Jaeyoung-Lim, 2023-07-05:
@arielc-brillianetor I don't think it is the same issue. Please post a new issue

Meowchine1, 2025-02-28:
Hi, I faced with the same problem, running on Ubuntu 22.04. But jmavsim works: ``` make px4_sitl jmavsim ```

Jaeyoung-Lim, 2025-02-28:
@Meowchine1 It is not the same problem if you are running a different Ubuntu. Please post a new issue