PX4/PX4-SITL_gazebo-classic #971: Satellite images not loading

jeremie-huy, 2023-03-29:
Hi,

I'm trying to get some satellite images from Google following this tutorial: https://classic.gazebosim.org/tutorials?tut=static_map_plugin&cat=build_world

It works fine when launching only Gazebo with this world, but when ran through PX4/Gazebo, it gets nothing (and the typhoon seems to be lost in the world).

Any advice?

Thanks

Jaeyoung-Lim, 2023-03-29:
> (and the typhoon seems to be lost in the world).

What does this mean? Please provide details of your setup, what is the expectation, what is not working, how it is not working. With the limited information you have provided, it is hard to know what might have gone wrong

jeremie-huy, 2023-03-30:
Sorry again, just realized that it was missing a `ground_plane`:
[image]

Anyway, now I can see that the satellite images are not there, here's the full world file:
```
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="default">
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <world_frame_orientation>ENU</world_frame_orientation>
      <latitude_deg>LAT</latitude_deg>
      <longitude_deg>LNG</longitude_deg>
      <elevation>0</elevation>
    </spherical_coordinates>

    <include>
      <uri>model://sun</uri>
    </include>
    <include>
      <uri>model://ground_plane</uri>
    </include>

    <physics name='default_physics' default='0' type='ode'>
      <gravity>0 0 -9.8066</gravity>
      <ode>
        <solver>
          <type>quick</type>
          <iters>10</iters>
          <sor>1.3</sor>
          <use_dynamic_moi_rescaling>0</use_dynamic_moi_rescaling>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>250</real_time_update_rate>
      <magnetic_field>6.0e-6 2.3e-5 -4.2e-5</magnetic_field>
    </physics>

    <plugin name="map" filename="libStaticMapPlugin.so">
      <center>LAT LNG</center>
      <world_size>100</world_size>
      <map_type>satellite</map_type>
      <api_key>API_KEY</api_key>
      <use_cache>false</use_cache>
    </plugin>
  </world>
</sdf>
```