PX4/PX4-SITL_gazebo-classic #972: Error: "[Err] [Plugin.hh:212] Failed to load plugin libgazebo_x.so: libgazebo_x.so: cannot open shared object file: No such file or directory"

Terence-I, 2023-04-03:
Hello! I have a question about the using the PX4-Autopilot package with the gazebo-classic simulator and ROS2 Foxy. In the github repository of the  PX4-Autopilot package under this path “PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/include” there are multiple gazebo plugins saved with the “.h” extension instead of the “.so” extension which gazebo plugins are usually saved with. When I run this command “./Tools/simulation/gazebo-classic/sitl_multiple_run.sh -t px4_sitl_rtps -m iris -n 2” to launch a gazebo simulation with two iris drone models, I get the error below which I think is due to the fact that the gazebo plugins used in the sdf file for the iris drone model have the “.so” extension instead of the “.h” extension.
Error:

“[Err] [Plugin.hh:212] Failed to load plugin libgazebo_gps_plugin.so: libgazebo_gps_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_multirotor_base_plugin.so: libgazebo_multirotor_base_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_groundtruth_plugin.so: libgazebo_groundtruth_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_magnetometer_plugin.so: libgazebo_magnetometer_plugin.so: cannot open shared object file: No such file or directory”

I am new to using gazebo with PX4-Autopilot and I was wondering if I need to compile these plugins into “.so” files, and if so how do I it? Or am I doing this wrong and I need to approach it differently? Any help regarding this would be extremely useful.

I am using ubuntu 20.04.6 LTS, ROS2 Foxy and Gazebo-11. Thank you!

Jaeyoung-Lim, 2023-04-03:
Which PX4 version?

You need to source your environment(
https://docs.px4.io/v1.13/en/simulation/multi_vehicle_simulation_gazebo.html#build-and-test)

```
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
```

Terence-I, 2023-04-03:
Thank you for a quick response, appreciate it. I tried your suggestion but I am still getting the error. I am using PX4 v1.13.3 (the latest version). In this version "setup_gazebo.bash" is stored under "/PX4-Autopilot/Tools/simulation/gazebo-classic/setup_gazebo.bash".

Terence-I, 2023-04-03:
Update: the error was resolved by the sourcing setup_gazebo.bash, I just forgot to also source .bashrc, after I did the error was resolved. 
Another question I had is how I can interact with the iris models in the gazebo simulation after running the "./Tools/simulation/gazebo-classic/sitl_multiple_run.sh -t px4_sitl_rtps -m iris -n 2" command. For example, are there terminal commands to tell the iris models to takeoff? are there ros topics I can work with to interact with the models? How would go about using these iris models in a custom gazebo world? Also, if there is any documentation that can help me find answers to these questions, it would be very helpful.

Again, thank you so much for your help!

mengchaoheng, 2024-05-28:
> 错误已通过获取 setup_gazebo.bash 解决，我只是忘了获取 .bashrc，

what is the detail? I have the same problem. @Terence-I