PX4/PX4-SITL_gazebo-classic #973: Logger fails to add topics

biagio-trimarchi, 2023-04-05:
I have recently downloaded the docker image "px4io/px4-dev-ros-noetic:latest".

When I run "make px4_sitl_default gazebo-classic" I get a lot of warning from the logger about "too many subscriptions".

I add here the complete log:

############################################################################

LOG 

#############################################################################

[0/4] Performing build step for 'sitl_gazebo-classic'
ninja: no work to do.
[3/4] cd /src/PX4-Autopilot/build/px4_sitl_default/src/modul.../src/PX4-Autopilot /src/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /src/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
model: iris
world: none
src_path: /src/PX4-Autopilot
build_path: /src/PX4-Autopilot/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic
GAZEBO_MODEL_PATH :/src/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models
LD_LIBRARY_PATH /opt/ros/noetic/lib:/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic
empty world, setting empty.world as default
Using: /src/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/iris/iris.sdf
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
SITL COMMAND: "/src/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/src/PX4-Autopilot/build/px4_sitl_default"/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 274 bytes, decoded 274 bytes (INT32:11, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 31: [: Illegal number: 
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO  [lockstep_scheduler] setting initial absolute time to 3388000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
Gazebo multi-robot simulator, version 11.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.200.15.240
[Wrn] [GuiIface.cc:298] Couldn't locate specified .ini. Creating file at "/home/user/.gazebo/gui.ini"
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
[... 72 lines cut ...]
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 7
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 8
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 9
WARN  [logger] Too many subscriptions, failed to add: collision_constraints 0
WARN  [logger] Too many subscriptions, failed to add: obstacle_distance_fused 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_mocap_odometry 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint_desired 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_visual_odometry 0
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2023-04-05/14_49_52.ulg	
ERROR [logger] Can't open log file ./log/2023-04-05/14_49_52.ulg, errno: 13
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] home set
[Wrn] [GuiIface.cc:120] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-user'
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
ERROR [flight_mode_manager] Failsafe task activated
INFO  [health_and_arming_checks] Preflight Fail: No manual control input	
INFO  [commander] Ready for takeoff!
INFO  [health_and_arming_checks] Preflight Fail: No manual control input

I

Jaeyoung-Lim, 2023-04-05:
@biagio-trimarchi Which px4 version are you using?

biagio-trimarchi, 2023-04-05:
I have downloaded the latest version of the docker container based on ROS Noetic.

px4io/px4-dev-ros-noetic:latest

I ran the container through the bash scripts:

# enable access to xhost from the container
xhost +

# Run docker and open bash shell
sudo docker run -it --privileged \
--env=LOCAL_USER_ID="$(id -u)" \
-v ~/PX4/PX4-Autopilot/:/src/PX4-Autopilot/:rw \
-v /tmp/.X11-unix:/tmp/.X11-unix:ro \
-e DISPLAY=$DISPLAY \
--network host \
--name=px4-ros px4io/px4-dev-ros-noetic:latest bash

Jaeyoung-Lim, 2023-04-06:
@biagio-trimarchi You are mapping a local volume containing PX4. Which px4 version is this?

biagio-trimarchi, 2023-04-06:
According to the package.xml inside the folder, it is 1.14.0

biagio-trimarchi, 2023-04-06:
I have just pulled from git the 1.13.2 version of PX4, with this previous version the logger does not rise any warnings.

julianoes, 2023-06-08:
As far as I'm aware this has been resolved in main and v1.14.

closed 2023-06-08 by julianoes