PX4/PX4-SITL_gazebo-classic #987: Issues while make command

kashifzr, 2023-05-27:
ROS : Noetic
Ubuntu : 20.04 Focal Fossa

Unable to Find issues while runing make command 

[ 98%] Built target gazebo_pose_sniffer_plugin
[ 98%] Built target gazebo_catapult_plugin
[ 98%] Built target gazebo_barometer_plugin
[ 98%] Built target gazebo_motor_model
[ 98%] Built target gazebo_wind_plugin
[ 98%] Built target gazebo_parachute_plugin
[ 98%] Built target gazebo_magnetometer_plugin
[ 98%] Built target AdvancedLiftDragPlugin
[ 98%] Built target gazebo_airship_dynamics_plugin
[ 98%] Built target ForceVisual
[ 98%] Built target LiftDragPlugin
[ 98%] Built target gazebo_gst_camera_plugin
[ 98%] Built target gazebo_drop_plugin
[ 99%] Built target gazebo_user_camera_plugin
[ 99%] Built target gazebo_video_stream_widget
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::targetReleativeCallback(gazebo::TargetRelativePtr&)’:
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:967:3: error: ‘mavlink_target_relative_t’ was not declared in this scope; did you mean ‘mavlink_parse_state_t’?
  967 |   mavlink_target_relative_t sensor_msg;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      |   mavlink_parse_state_t
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:968:3: error: ‘sensor_msg’ was not declared in this scope; did you mean ‘sensor_msgs’?
  968 |   sensor_msg.timestamp = targetRelative_message->time_usec();
      |   ^~~~~~~~~~
      |   sensor_msgs
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:974:22: error: ‘TARGET_OBS_FRAME_BODY_FRD’ was not declared in this scope; did you mean ‘MAV_FRAME_BODY_FRD’?
  974 |   sensor_msg.frame = TARGET_OBS_FRAME_BODY_FRD;
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                      MAV_FRAME_BODY_FRD
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:993:3: error: ‘mavlink_msg_target_relative_encode_chan’ was not declared in this scope; did you mean ‘mavlink_msg_heartbeat_encode_chan’?
  993 |   mavlink_msg_target_relative_encode_chan(1, 200, MAVLINK_COMM_0, &msg, &sensor_msg);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   mavlink_msg_heartbeat_encode_chan
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::TargetGpsCallback(gazebo::GpsPtr&)’:
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:999:3: error: ‘mavlink_target_absolute_t’ was not declared in this scope; did you mean ‘mavlink_parse_state_t’?
  999 |   mavlink_target_absolute_t gps_data;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      |   mavlink_parse_state_t
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1001:3: error: ‘gps_data’ was not declared in this scope; did you mean ‘epoll_data’?
 1001 |   gps_data.timestamp = gps_msg->time_usec();
      |   ^~~~~~~~
      |   epoll_data
/home/rob-kkj/home/PX4/src/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1019:3: error: ‘mavlink_msg_target_absolute_encode_chan’ was not declared in this scope; did you mean ‘mavlink_msg_param_value_encode_chan’?
 1019 |   mavlink_msg_target_absolute_encode_chan(1, 200, MAVLINK_COMM_0, &msg, &gps_data);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   mavlink_msg_param_value_encode_chan
make[2]: *** [CMakeFiles/gazebo_mavlink_interface.dir/build.make:76: CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:855: CMakeFiles/gazebo_mavlink_interface.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

paulerikf, 2023-12-20:
Did you ever figure this out?

kashifzr, 2023-12-20:
@paulerikf thanks you reply
i figure my issue way back by using proper source code for PX4. I am using this package for while but unable to understand when i used for second time, i never see this issue before