PX4/PX4-SITL_gazebo-classic #991: Cannot get data of cameras when simulating multiple iris drone

longvucao, 2023-06-10:
Hi, I am trying to simulate multiple iris drone and want to get its camera at the same time. When I worked with a drone, I added the following code to the end of the iris.sdf file to get the camera:
[image]

Then I used the file mavros_posix_sitl.launch and it got the image I wanted (you can see it in the picture)
[image]
[image]

However when I use the file multi_uav_mavros_sitl.launch to simulate multiple drones (still using the edited iris.sdf file) the desired topic appears but I can't get the image (you can see the difference here picture below)
[image]

So do you have any ideas to solve this problem?
I really appreciate your help)

JohnTGZ, 2023-09-07:
Did you manage to find a way to namespace all the different camera topics being published so that they are distinct from each other? i.e. not publishing to the same camera topic.

Babylonehy, 2023-10-31:
Use group namesapce when you spawn your model into gazebo. You have to make sure your depth camera have <robot_namespace> </robot_namespace>.

mun0404, 2024-12-03:
@Babylonehy I have tried using group namespace but I am still getting the namespace error.

LohitoBurrito, 2025-02-27:
@mun0404 did you end up figuring this out, I am really stuck. I am about to create 2 new separate iris drone models with different namespace for the cameras

mun0404, 2025-03-04:
@LohitoBurrito No, I couldn't figure it out. I am still trying to figure out a solution for it, please let me know if you arrive at a solution.