PX4/PX4-SITL_gazebo-classic #1001: gazebo_motor_model.cpp , what is the unit for the motor_constant  and where is the moment calculated?

mariam0amin, 2023-09-06:
ignition::math::Vector3d drag_torque(0, 0, -turning_direction_ * force * moment_constant_); 
is this the calculation for the moment? and if it is, how is it force* moment constant, does this mean moment constant unit is meters??

JohnTGZ, 2023-09-07:
Based on [these notes](https://github.com/mvernacc/gazebo_motor_model_docs/blob/master/notes.pdf) which try to explain the gazebo parameters for the motor model, you might actually be right.

mariam0amin, 2023-09-10:
@JohnTGZ  
Thank you so much,but it is also written that the motor_constant is in kg.m in this part of the paper 
[image]
 however I thought it was in (N.sec^2/rad^2)   on what I thought I understand and also based on these two issues:
[5986](https://discuss.px4.io/t/thrust-and-torque-constants-in-toml-geometry-description-and-sitl-gazebo-models/5986) 
[110](https://github.com/PX4/PX4-SITL_gazebo-classic/issues/110)
so how can the motor constant have this unit?

JohnTGZ, 2023-09-11:
Motor constant is in (kg)(m).
Moment constant is in (m)

According to the (code for motor constant and force)[https://github.com/PX4/PX4-SITL_gazebo-classic/blob/d08eb5b22d3edfb7a740c883b06f895a173b5519/src/gazebo_motor_model.cpp#L194]:
Motor Constant = Thrust / (Angular_rate^2) 
Units of motor constant = [Mass][Length][Time^2] / [Time^2] = [Mass][Length]

According to the (code for moment constant and torque)[https://github.com/PX4/PX4-SITL_gazebo-classic/blob/d08eb5b22d3edfb7a740c883b06f895a173b5519/src/gazebo_motor_model.cpp#L233]:
Moment Constant = (Torque about z axis) / (Thrust along z axis) 
Units of Moment constant = [Mass][Length^2][Time^2] / [Mass][Length][Time^2] = [Length]