PX4/PX4-SITL_gazebo-classic #1003: I can't compile it in ubuntu 22.04

qfwysw, 2023-09-11:
I try to install ros-humble-mavlink, but it does not work. So what should I do? 
qf@qf-Lenovo-Legion-Y7000:~/px4_sitl_gazebo/src/sitl_gazebo/build$ cmake ..
-- install-prefix: /usr
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.58") found components: system thread filesystem 
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  CMakeLists.txt:48 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Looking for ignition-math6 -- found version 6.12.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.12.0
-- Looking for ignition-transport8 -- found version 8.2.1
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found suitable version "3.12.4", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.12.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Looking for ignition-common3 -- found version 3.14.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.14.0
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.12.0
-- Looking for ignition-fuel_tools4 -- found version 4.4.0
-- Searching for dependencies of ignition-fuel_tools4
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.5
-- Checking for module 'libzip'
--   Found libzip, version 1.7.3
-- Looking for ignition-common3 -- found version 3.14.0
-- Looking for ignition-math6 -- found version 6.12.0
-- Looking for ignition-msgs5 -- found version 5.8.1
CMake Deprecation Warning at external/OpticalFlow/CMakeLists.txt:34 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
[... 285 lines cut ...]
CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "gazebo_user_camera_plugin":

    "_MAVLINK_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "LiftDragPlugin":

    "_MAVLINK_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of "ForceVisual":

    "_MAVLINK_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "AdvancedLiftDragPlugin":

    "_MAVLINK_INCLUDE_DIR-NOTFOUND"

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

tokilligent, 2023-09-12:
I am facing the same issue and couldn't find a solution as of now.

hamishwillee, 2023-09-12:
@Jaeyoung-Lim Anything you can help with?

tokilligent, 2023-09-18:
So the solution to the above problem is, you need to generate header files from the `mavlink` folder. After that, you need to put those files into the `sitl_gazebo` directory.

closed 2024-06-15 by Jaeyoung-Lim

julianoes, 2025-03-13:
For anyone finding this in the future, the solution is over here: https://github.com/PX4/PX4-Autopilot/issues/20632#issuecomment-1483748872