PX4/PX4-SITL_gazebo-classic #1004: Multi-rotor + Motor with angle

Osama-Mamdouh, 2023-09-11:
I have a multi-rotor drone and I've set up the model and run it on PX4. Now, I need to make some modifications to make the motors tilt at an angle on the arms, and I know this will affect some files. Can anyone tell me how to do that?

Jaeyoung-Lim, 2023-09-18:
@Osama-Mamdouh Have you checked the omnicopter model?

Osama-Mamdouh, 2023-09-18:
> @Osama-Mamdouh Have you checked the omnicopter model?

No I didn't do that.
Does this model exist within px4 Autopilot ?
Note: I am using release 1.11.
Thank you very much,

Jaeyoung-Lim, 2023-09-18:
@Osama-Mamdouh It doesn't exist in v1.11, but you can find it an up to date version of this repo

Osama-Mamdouh, 2023-09-21:
> @Osama-Mamdouh It doesn't exist in v1.11, but you can use it as an inspiration.
> 
> You can find it an up to date version of this repo

Thank you very much for your help

closed 2023-09-21 by Jaeyoung-Lim