PX4/PX4-SITL_gazebo-classic #1005: "Multirotor simulation using sitl px4 on gazebo"how to add tilt angle to the model?

mariam0amin, 2023-09-17:
hello everyone. I have a multicopter with all of the rotors tilted with an angle on the arms, so how can I add the effect of the tilt angle of the rotors? Is it only through the mixer, or is there a place or a file where I should add its value?

Jaeyoung-Lim, 2023-09-17:
@mariam0amin The mixer or control allocation (depending on which PX4 version you are using) affects the autopilot firmware, not the simulation. Therefore you would need to define the correct actuator configurations on the PX4 side and also create a gazebo model that corresponds to your model.

For the gazebo side, you would need to define your model as done for other models: https://github.com/PX4/PX4-SITL_gazebo-classic/blob/main/models/iris/iris.sdf.jinja

If you want to learn more about how to define sdf files: https://classic.gazebosim.org/tutorials?tut=build_model

mariam0amin, 2023-09-17:
@Jaeyoung-Lim  thank you so much, I understand that modifying the mixer would not affect the simulation and I am okay with modeling on gazebo side. but for the px4 side, I am using Px4 v1.11 , so how can I define the correct actuator configuration in the px4?  and  how can I show the tilt angle effect through the mixer "I mean how to show its effect in the dynamics of the multicopter and its behaviour change due to this tilt"?
thanks again in advance

Jaeyoung-Lim, 2023-09-17:
You just need to define the mixer in a way that it represents your input. This you would need to calculate if you are using a mixer. For more recent versions of px4, you can simply put the axis directions of the motor, and it would automatically account for it.
- https://docs.px4.io/main/en/config/actuators.html

> I mean how to show its effect in the dynamics of the multicopter and its behaviour change due to this tilt"?

This is a completely different question. what behavior change are you expecting?

Note that there already exists an omnicopter model which has fixed tilt: https://github.com/Jaeyoung-Lim/px4-manipulation

mariam0amin, 2023-09-18:
@Jaeyoung-Lim  thank you so much for your help, is version 1.11 considered an old version? if so where should I modify the mixer, would I just add the geometry file for my vehicle and specify the position of each rotor through this path:
 https://github.com/PX4/PX4-Autopilot/tree/release/1.11/src/lib/mixer/MultirotorMixer/geometries
or there is another thing to do to account for it? (because I don't understand how would this account for the tilt angle)
and for the second part of my question, sorry I did not mean to ask a different question, I just couldn't explain what I meant clearly.

Jaeyoung-Lim, 2023-09-18:
@mariam0amin You can check the released versions here: https://github.com/PX4/PX4-Autopilot/releases, v1.11 was released in 2020

You can have a look on how to configure your actuators using mixers here: https://dev.px4.io/v1.11_noredirect/en/concept/mixing.html

mariam0amin, 2023-09-18:
@Jaeyoung-Lim Great, Thank you so much.

closed 2023-09-18 by Jaeyoung-Lim