PX4/PX4-SITL_gazebo-classic #1006: Unable to design Custom Drone with velodyne Sensor

kashifzr, 2023-09-18:
I work around to design sdf model for Velodyne with Iris drone using iris_rplidar as reference from models, when I run the mavros_posix_sitl.launch file with vehicle := iris_velodyne then output is like 

<pre>auto-starting new master
<b>process[master]: started with pid [156819]</b>
<b>ROS_MASTER_URI=http://localhost:11311</b>

<b>setting /run_id to b962af52-55ce-11ee-be1a-37316987660f</b>
<b>process[rosout-1]: started with pid [156844]</b>
started core service [/rosout]
<b>process[sitl-2]: started with pid [156852]</b>

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  <font color="#D3D7CF">[px4] </font>startup script: /bin/sh etc/init.d-posix/rcS 0
<b>process[gazebo-3]: started with pid [156863]</b>
ERROR [init] Unknown model gazebo-classic_iris_velodyne (not found by name on /home/rob-kkj/.ros/etc/init.d-posix/airframes)
<font color="#CC0000">ERROR </font><font color="#D3D7CF">[px4] </font><font color="#CC0000">Startup script returned with return value: 256</font>
<b>process[gazebo_gui-4]: started with pid [156872]</b>
<b>process[vehicle_spawn_robkkj_System_Product_Name_156796_3537816507072250960-5]: started with pid [156877]</b>
<b>process[mavros-6]: started with pid [156879]</b>
<font color="#CC0000">================================================================================REQUIRED process [sitl-2] has died!</font>
<font color="#CC0000">process has died [pid 156852, exit code 255, cmd /home/rob-kkj/home/PX4/Firmware/build/px4_sitl_default/bin/px4 /home/rob-kkj/home/PX4/Firmware/build/px4_sitl_default/etc -s etc/init.d-posix/rcS __name:=sitl __log:=/home/rob-kkj/.ros/log/b962af52-55ce-11ee-be1a-37316987660f/sitl-2.log].</font>
<font color="#CC0000">log file: /home/rob-kkj/.ros/log/b962af52-55ce-11ee-be1a-37316987660f/sitl-2*.log</font>
<font color="#CC0000">Initiating shutdown!</font>
<font color="#CC0000">================================================================================</font>
[ INFO] [1695005686.270737354]: FCU URL: udp://:14540@localhost:14557
[ INFO] [1695005686.272585668]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1695005686.272675346]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1695005686.272718579]: GCS bridge disabled
[ INFO] [1695005686.279309973]: Plugin 3dr_radio loaded
[ INFO] [1695005686.280324646]: Plugin 3dr_radio initialized
[ INFO] [1695005686.280390717]: Plugin actuator_control loaded
[ INFO] [1695005686.281897990]: Plugin actuator_control initialized
[ INFO] [1695005686.284810565]: Plugin adsb loaded
[ INFO] [1695005686.286266713]: Plugin adsb initialized
[ INFO] [1695005686.286352061]: Plugin altitude loaded
[ INFO] [1695005686.286794374]: Plugin altitude initialized
[ INFO] [1695005686.286858699]: Plugin cam_imu_sync loaded
[ INFO] [1695005686.287135275]: Plugin cam_imu_sync initialized
[ INFO] [1695005686.287196318]: Plugin camera loaded
[ INFO] [1695005686.287497688]: Plugin camera initialized
[ INFO] [1695005686.287562293]: Plugin cellular_status loaded
[ INFO] [1695005686.288650231]: Plugin cellular_status initialized
[ INFO] [1695005686.288713089]: Plugin command loaded
[ INFO] [1695005686.291499599]: Plugin command initialized
[ INFO] [1695005686.291570140]: Plugin companion_process_status loaded
[ INFO] [1695005686.292757324]: Plugin companion_process_status initialized
[ INFO] [1695005686.292823674]: Plugin debug_value loaded
[ INFO] [1695005686.295359449]: Plugin debug_value initialized
[ INFO] [1695005686.295395697]: Plugin distance_sensor blacklisted
[ INFO] [1695005686.295462746]: Plugin esc_status loaded
[ INFO] [1695005686.296032871]: Plugin esc_status initialized
[ INFO] [1695005686.296103831]: Plugin esc_telemetry loaded
[ INFO] [1695005686.296403386]: Plugin esc_telemetry initialized
[ INFO] [1695005686.296481261]: Plugin fake_gps loaded
[ INFO] [1695005686.304021955]: Plugin fake_gps initialized
[ INFO] [1695005686.304127487]: Plugin ftp loaded
[ INFO] [1695005686.310500064]: Plugin ftp initialized
[ INFO] [1695005686.310609926]: Plugin geofence loaded
[ INFO] [1695005686.312157428]: Plugin geofence initialized
[ INFO] [1695005686.312249900]: Plugin global_position loaded
[ INFO] [1695005686.320160062]: Plugin global_position initialized
[ INFO] [1695005686.320272578]: Plugin gps_input loaded
[... 68 lines cut ...]
[ INFO] [1695005686.359117347]: Plugin odom initialized
[ INFO] [1695005686.359190192]: Plugin onboard_computer_status loaded
[ INFO] [1695005686.360095562]: Plugin onboard_computer_status initialized
[ INFO] [1695005686.360166242]: Plugin param loaded
[ INFO] [1695005686.361323255]: Plugin param initialized
[ INFO] [1695005686.361381015]: Plugin play_tune loaded
[ INFO] [1695005686.362411682]: Plugin play_tune initialized
[ INFO] [1695005686.362476495]: Plugin px4flow loaded
[ INFO] [1695005686.365267754]: Plugin px4flow initialized
[ INFO] [1695005686.365343254]: Plugin rallypoint loaded
[ INFO] [1695005686.366497962]: Plugin rallypoint initialized
[ INFO] [1695005686.366520311]: Plugin rangefinder blacklisted
[ INFO] [1695005686.366582331]: Plugin rc_io loaded
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
<b>done</b>
</pre>

OS = 20.04 Focal and ROS = Noetic. 

in sdf file for velodyne sensor  i am using plugin = libgazebo_ros_velodyne_laser.so

unable to find how to add custom models for sensors and drone are being integrated in PX4-Autopilot

Jaeyoung-Lim, 2023-09-18:
> ERROR [px4] Startup script returned with return value: 256

@kashifzr You also need a airframe config for the vehicle on the PX4 side that matches the model name. Or in the launchscript you can set different sdf models and airframes independently.

kashifzr, 2023-09-18:
@Jaeyoung-Lim thanks for giving me idea to solve this issue.

 I see some hints specific to airframe   #855. 
on https://docs.px4.io/main/en/dev_airframes/adding_a_new_frame.html give information specific to frame configuration 
I do not understand all the details i so i will check the details

How to add air frames in launch script Any Idea ?

sycamore-yu, 2024-10-20:
I meet the same problem and finally figure it out!!!!!
in your launch.file ,don not chang 
`<arg name="vehicle" default="iris"/>`
if u use 
`<arg name="vehicle" default="iris_D435i"/> `
or other models,the PX4 init do not recogonize it and then 
```
ERROR [init] Unknown model gazebo-classic_iris_velodyne (not found by name on /home/rob-kkj/.ros/etc/init.d-posix/airframes)
ERROR [px4] Startup script returned with return value: 256

```
add your sdf file below like this
```
  <!-- vehicle model and world -->
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_D435i/iris_D435i.sdf"/>
```

it works for me

sycamore-yu, 2024-10-20:
@kashifzr hope be helpful