PX4/PX4-SITL_gazebo-classic #1008: I want to use Gazebo to emulate Caipiroshka Duo Tailsitter

waitfordark, 2023-09-18:
At present, Gazebo can simulate tailsitter, and I want to simulate caipiroshka duo tailsitter through gazebo. During the current simulation, caipiroshka duo tailsitter always has attitude confusion when taking off. Which led to the crash. After comparing the caipiroshka duo tailsitter, it is found that Caipiroshka Duo tailsitter needs two rudder surfaces for attitude control during takeoff. I guess it is because my rudder model is not accurate enough, which leads to the takeoff failure, but I still can't adjust it. I would like to ask you to give me some suggestions. 

Thank you!

Userpc1010, 2023-09-18:
Yes, and is it possible to somehow add tilt rotors for hover mode?)))

waitfordark, 2023-09-19:
Thank you for your reply！

I don't think that's quite right. From the PX4 source code analysis, the difference between Caipiroshka Duo Tailsitter and Tailsitter is that the tailsitter only needs 4 rotor motors to take off smoothly during takeoff. Besides the two rotor motors, two rudder motors are also needed to maintain the attitude balance. The modeling effect of Caipiroshka Duo Tailsitter's rudder model is not good, but I do not know how to improve it. I don't think that's quite right. From the PX4 source code analysis, the difference between Caipiroshka Duo Tailsitter and Tailsitter is that the tailsitter only needs 4 rotor motors to take off smoothly during takeoff. Besides the two rotor motors, two rudder motors are also needed to maintain the attitude balance. The modeling effect of Caipiroshka Duo Tailsitter's rudder model is not good, but I do not know how to improve it.

Userpc1010, 2023-09-19:
My idea is to replace the elevons with a servo drive that tilts the rotor according to one axis, pitch (roll & yaw we have in 2 rotos on hover mode) for hanging in a vertical position perhaps this will require a non-standard airframe and also add 2 servo for the elevons in horizont fly.  It is difficult to simulate the moments occurring at the elevons because gazebo models thrust as a force and not a stream of particles reflected from the rudders, but I think it’s possible to rotate the thrust vector of the engines themselves along one axis of pitch? In the gazebo there is an example of tiltrotor, you need to combine them with the example of a tailsitter on two motors if the angular position of the rotors can be controlled, you get a thrust vector.

hamishwillee, 2023-09-19:
@Jaeyoung-Lim Can you provide a starting point for this, or is it something that needs to be addressed first though the vtol discussion boards and perhaps with Silvan/Roman?

waitfordark, 2023-09-20:
@Jaeyoung-Lim  Yes, we're focusing on Duo tailsitter. I have adjusted the elevon model and model parameters in the last two days, and now it can hover stably. Thank you for your attention! @hamishwillee @Userpc1010

Userpc1010, 2023-09-21:
I did some tests with my model with two rotors and ran into a problem with the signals from the elevons freezing during vertical takeoff. I also tried to reconfigure the airframe manually in QGC (using Caipiroshka Duo Tailsitter) but got the same result the PWM outputs after the start freeze at position 1000 or 1500 without controlling the pitch(((.

[image]

Although in horizontal flight they try to work:

[image]

https://youtu.be/Bte9bHw2Rpo

My model:

[Dual_Rotor_tailsitter1.zip](https://github.com/PX4/PX4-SITL_gazebo-classic/files/12686734/Dual_Rotor_tailsitter1.zip)

Tilt rotor not correct to:

[image]
[image]

waitfordark, 2023-09-22:
@Userpc1010 
First, you need to ensure that the mixing channel is correct. According to the definition of duo tailsitter mixing document, channel 1 and channel 2 correspond to rotor motors, which mainly provide propeller pull during takeoff to make the aircraft leave the ground, while channel 5 and channel 6 correspond to rudder motors, which are used to control the left and right elevon to control the attitude during takeoff.

Secondly, it should be noted that in the simulation process, the input control amount of the rudder motor is positive and negative.

At the same time, the installation position of the motor I set is above the center of gravity of the aircraft, and the installation position of the steering gear is below the center of gravity of the aircraft.

I hope my reply can help you!

mengchaoheng, 2026-01-08:
@Jaeyoung-Lim @waitfordark @Userpc1010 
I am building a simulation of a tail-sitter VTOL with two rotors, and I plan to push the PR later. Currently, the liftdrag_plugin is running well, while the advanced_liftdrag_plugin still needs further improvement.

[PR](https://github.com/PX4/PX4-SITL_gazebo-classic/pull/1083) is in progress.