PX4/PX4-SITL_gazebo-classic #1013: Compatibility with opencv >4.6

danielhonies, 2023-10-04:
Starting with Opencv 4.7 this dictionary.hpp does not exist any longer because the aruco module was restructured.

line 42 #include <opencv2/aruco/dictionary.hpp> in https://github.com/PX4/PX4-SITL_gazebo-classic/blob/main/include/gazebo_aruco_plugin.h

This is a problem when compiling on macOS as brew installs opencv 4.8.1 on default and it's only possible to get it to work by manually compiling opencv.

wntun, 2023-10-17:
Yes, I also had the same problem but fixed the codes. It's working now. 
I hope it's helpful.
[gazebo_aruco_plugin.cpp.txt](https://github.com/PX4/PX4-SITL_gazebo-classic/files/12923417/gazebo_aruco_plugin.cpp.txt)
[gazebo_aruco_plugin.h.txt](https://github.com/PX4/PX4-SITL_gazebo-classic/files/12923420/gazebo_aruco_plugin.h.txt)

danielhonies, 2023-10-17:
Thanks, I created a PR with the changes

closed 2023-11-14 by Jaeyoung-Lim