PX4/PX4-SITL_gazebo-classic #1016: Gazebo Classic PX4 SITL Multi-vehicle Real Time Slows Down.

eng-86, 2023-10-17:
Dear, I am working on multi-vehicle SITL PX4 in gazebo with ROS. When running four iris quads the real time in Gazebo slows down to 0.2, only when launching the simulation initially. Also I did not run any scrips, any task is performed, no take off is done, no ROS package was launched, no Qground was connected....
Please note that when running single vehicle with ROS, the real time is 1. I checked the CPU and RAM resources in my ubuntu, all were fine (<25%).
Please help

danielhonies, 2023-11-09:
Does the sim time go up after the initial slow down? Could be normal

Jaeyoung-Lim, 2023-11-09:
@eng-86 How are you starting the simulations? I could run 200 vehicles with realtime factor of 0.9, so I don't think this is normal

eng-86, 2023-11-09:
@danielhonies @Jaeyoung-Lim The simulation starts with 0.2 factor and continues.

eng-86, 2023-11-09:
I am a PhD student. I am working on a distributive fault tolerant control for multi-UAVs. Each UAV can estimate its local actuator fault and neighbors fault relying on torques and thrust control inputs to each UAV in addition to UAVs' states. I was using a one-year-recent version PX4 with ROS to verify my proposed algorithms. It was working except that the real time scale in GAZEBO persists at 0.2 when trying to simulate four vehicles with an error "ERROR [param] Parameter XRCE_DDS_KEY not found.". This caused an accuracy problem (the algorithm was also validated through matlab simulation). When I upgraded the code to the latest one, the real time problem was solved, but a new problem appeared. My algorithm depends on desired mavros actuator target controls for each UAV to estimate the actuator fault, and it is not published in new versions (I checked it during simulation and according to the discussion also https://github.com/mavlink/mavlink/pull/1784 ). I compared also two log data samples. I found that in old version, the actuator control log was found (Roll, Pitch, Yaw, Thrust (up), Thrust (forward).

eng-86, 2023-11-09:
Additional information: I returned to PX4 13.3 release, I tested 4 UAVs with GAZEBO and ROS with same python script used before. Time factor is 1, but an additional problem appeared: when I did some changes to the code and built it, the changes did not reflect on the results (in other words, no effect of changes). The changes are injecting actuator fault after a time from takeoff. This is not the first time I perform such changes.

Jaeyoung-Lim, 2023-11-09:
@eng-86 I was asking how you are starting the sim(which script/launchfile)

eng-86, 2023-11-09:
@Jaeyoung-Lim I started through launch file as follows:
cd PX4-Autopilot
git submodule update --init --recursive
DONT_RUN=1 make px4_sitl_default gazebo-classic
source Tools/simulation/gazebo-classic/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic
roslaunch px4 multi_uav_mavros_sitl.launch

even before arming and starting any script,the time factor is 0.2

eng-86, 2023-11-09:
I am struggling with this issue, and it is important  to publish my paper. I validated my work through MATLAB, and since I am planning to publish in a good journal (IEEE transaction) they need some real validation.

Jaeyoung-Lim, 2023-11-10:
@Jaeyoung-Lim Have you tried running the multivehicle script directly? https://github.com/PX4/PX4-Autopilot/blob/main/Tools/simulation/gazebo-classic/sitl_multiple_run.sh

hamishwillee, 2023-11-15:
Is this resolved @eng-86 ?

eng-86, 2023-11-15:
@hamishwillee  @Jaeyoung-Lim , I currently returned to PX4 version v13.3 since as I mentioned in https://github.com/mavlink/mavlink/pull/1784#issuecomment-1774460620 and https://github.com/PX4/PX4-Autopilot/issues/22265#issue-1962761660 , the  ACTUATOR_CONTROL_TARGET message is not published in newer versions  and I need it in my research. In v13.3 the simulation is going fine. Exactly in v14.0 beta version, it is published but the simulation time factor is 0.2. After this version, it is not published. I am working on this issue, it is part of my research, and I will give the update when it is ready. Thank you.