PX4/PX4-SITL_gazebo-classic #1020: Encounter errors after the homebrew auto upgraded

kevinuav, 2023-11-01:
I am using OSX(v12.6.6 Monterey) to run the simulations. It works well until the home-brew auto upgraded some of the formulas. How can I fix the issues?

The errors:

 PX4-Autopilot % make px4_sitl gazebo_plane
[0/1] cd /Users/kevin/PX4-Autopilot/bu...4_sitl_default -- gazebo-classic_plane
WARNING gazebo_plane target DEPRECATED, please use gazebo-classic_plane
[0/4] Performing build step for 'sitl_gazebo-classic'
[1/15] Building CXX object CMakeFiles/...e/moc_gazebo_video_stream_widget.cpp.o
FAILED: CMakeFiles/gazebo_video_stream_widget.dir/include/moc_gazebo_video_stream_widget.cpp.o 
/usr/local/bin/ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DLIBBULLET_VERSION=3.25 -DLIBBULLET_VERSION_GT_282 -DPROTOBUF_USE_DLLS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DSPDLOG_FMT_EXTERNAL -Dgazebo_video_stream_widget_EXPORTS -I/Users/kevin/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/include -I/Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -I/usr/local/include/eigen3/eigen3 -I/usr/local/include/gazebo-11/gazebo/msgs -I/Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink -I/Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink/mavlink/v2.0 -I/usr/local/include/Paging -I/Users/kevin/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/external/OpticalFlow/include -I/usr/local/Cellar/gstreamer/1.22.6/include/gstreamer-1.0 -I/usr/local/Cellar/glib/2.78.1/include -I/usr/local/Cellar/glib/2.78.1/include/glib-2.0 -I/usr/local/Cellar/glib/2.78.1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre2/10.42/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/ffi -isystem /usr/local/include -isystem /usr/local/include/gazebo-11 -isystem /usr/local/Cellar/bullet/3.25/include/bullet -isystem /usr/local/Cellar/bullet/3.25/include -isystem /usr/local/include/simbody -isystem /usr/local/Cellar/protobuf/24.4/include -isystem /usr/local/Cellar/abseil/20230802.1/include -isystem /usr/local/include/sdformat-9.10 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Paging -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /usr/local/opt/qt@5/lib -isystem /usr/local/opt/qt@5/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt@5/./mkspecs/macx-clang -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/opencv/4.8.1_1/include/opencv4 -isystem /usr/local/Cellar/assimp/5.3.1/include -isystem /usr/local/include/sdformat-9.10/sdf/.. -isystem /usr/local/include/ignition/cmake2 -isystem /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers -isystem /usr/local/opt/qt@5/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers -O2 -g -DNDEBUG -std=gnu++17 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -mmacosx-version-min=12.6 -fPIC -F/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/System/Library/Frameworks  -Wno-deprecated-declarations -Wno-address-of-packed-member -mfpmath=sse -msse -msse2 -msse3 -mssse3 -fPIC -MD -MT CMakeFiles/gazebo_video_stream_widget.dir/include/moc_gazebo_video_stream_widget.cpp.o -MF CMakeFiles/gazebo_video_stream_widget.dir/include/moc_gazebo_video_stream_widget.cpp.o.d -o CMakeFiles/gazebo_video_stream_widget.dir/include/moc_gazebo_video_stream_widget.cpp.o -c /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/moc_gazebo_video_stream_widget.cpp
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/moc_gazebo_video_stream_widget.cpp:10:
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/../../../../Tools/simulation/gazebo-classic/sitl_gazebo-classic/include/gazebo_video_stream_widget.h:21:
In file included from /usr/local/include/gazebo-11/gazebo/gui/GuiPlugin.hh:20:
In file included from /usr/local/include/gazebo-11/gazebo/gui/qt.h:39:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/QAction:1:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/qaction.h:46:
/usr/local/include/QtWidgets/qwidget.h:576:26: error: incomplete type 'QAction' named in nested name specifier
        connect(result, &QAction::triggered, std::forward<Args>(args)...);
                         ^~~~~~~~~
/usr/local/include/QtGui/qevent.h:35:7: note: forward declaration of 'QAction'
class QAction;
      ^
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/moc_gazebo_video_stream_widget.cpp:10:
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/../../../../Tools/simulation/gazebo-classic/sitl_gazebo-classic/include/gazebo_video_stream_widget.h:21:
In file included from /usr/local/include/gazebo-11/gazebo/gui/GuiPlugin.hh:20:
In file included from /usr/local/include/gazebo-11/gazebo/gui/qt.h:39:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/QAction:1:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/qaction.h:46:
/usr/local/include/QtWidgets/qwidget.h:583:26: error: incomplete type 'QAction' named in nested name specifier
        connect(result, &QAction::triggered, std::forward<Args>(args)...);
                         ^~~~~~~~~
/usr/local/include/QtGui/qevent.h:35:7: note: forward declaration of 'QAction'
class QAction;
      ^
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/moc_gazebo_video_stream_widget.cpp:10:
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/../../../../Tools/simulation/gazebo-classic/sitl_gazebo-classic/include/gazebo_video_stream_widget.h:21:
In file included from /usr/local/include/gazebo-11/gazebo/gui/GuiPlugin.hh:20:
In file included from /usr/local/include/gazebo-11/gazebo/gui/qt.h:39:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/QAction:1:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/qaction.h:46:
/usr/local/include/QtWidgets/qwidget.h:601:26: error: incomplete type 'QAction' named in nested name specifier
        connect(result, &QAction::triggered, std::forward<Args>(args)...);
                         ^~~~~~~~~
/usr/local/include/QtGui/qevent.h:35:7: note: forward declaration of 'QAction'
class QAction;
      ^
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/moc_gazebo_video_stream_widget.cpp:10:
In file included from /Users/kevin/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/include/../../../../Tools/simulation/gazebo-classic/sitl_gazebo-classic/include/gazebo_video_stream_widget.h:21:
In file included from /usr/local/include/gazebo-11/gazebo/gui/GuiPlugin.hh:20:
In file included from /usr/local/include/gazebo-11/gazebo/gui/qt.h:39:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/QAction:1:
In file included from /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers/qaction.h:46:
/usr/local/include/QtWidgets/qwidget.h:608:26: error: incomplete type 'QAction' named in nested name specifier
        connect(result, &QAction::triggered, std::forward<Args>(args)...);
[... cut ...]
FAILED: src/modules/simulation/simulator_mavlink/CMakeFiles/gazebo_plane /Users/kevin/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/simulator_mavlink/CMakeFiles/gazebo_plane 
cd /Users/kevin/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/simulator_mavlink && /usr/local/Cellar/cmake/3.27.7/bin/cmake -E cmake_echo_color --red "WARNING gazebo_plane target DEPRECATED, please use gazebo-classic_plane" && /usr/local/Cellar/cmake/3.27.7/bin/cmake --build /Users/kevin/PX4-Autopilot/build/px4_sitl_default -- gazebo-classic_plane
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

danielhonies, 2023-11-09:
Seems QT related. Check which version of qt is active in brew and try to switch it to an older one

ryanjAA, 2023-11-12:
Same issue. As mentioned above, uninstalled non @5 qt and pyqt resolves.

hamishwillee, 2023-11-15:
@danielhonies Is this fixed by your recent patch PR?

danielhonies, 2023-11-16:
@hamishwillee it is not. Unfortunately there are a bunch of issues with macOS and haven't had the time to look into all of them

ryanjAA, 2023-11-16:
Ya - removing the non @ qt is messing with me building normal firmware...

hamishwillee, 2023-11-16:
Thanks @danielhonies -  just wishful thinking really :-). Pleased to see at least some work going on to support this OS.