PX4/PX4-SITL_gazebo-classic #1021: Build issues on macOS Sonoma 14.1

ryanjAA, 2023-11-16:
Fresh install of macOS Sonoma 14.1

Fresh install of PX4 toolchain and related.

Clone of Main (also tried RC2 because I had it pulled).

Gazebo (classic) starts fine (just invoking Gazebo)

I also point the Xcode sdk to version 12.1 afterwards, see very bottom via:
(CMAKE_OSX_SYSROOT "/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk")  in the cmake (no luck either).

I went ahead and tried this on release I get a whole bunch of set but errors which you can get through by adding voids to the empty paranthesis but you still end up with the same errors seen here..

So, with no changes to anything I get this:

```
[0/954] git submodule src/drivers/gps/devices
[8/954] git submodule src/modules/mavlink/mavlink
[10/954] git submodule Tools/simulation/gazebo-classic/sitl_gazebo-classic
[11/954] git submodule src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client
[18/954] Performing configure step for 'sitl_gazebo-classic'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.58") found components: system thread filesystem 
-- Found DART: /usr/local/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Looking for ignition-math6 -- found version 6.15.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib
-- Looking for ignition-math6 -- found version 6.15.0
-- Looking for ignition-transport8 -- found version 8.4.0
-- Searching for dependencies of ignition-transport8
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.5
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.15.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Looking for ignition-common3 -- found version 3.16.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.16.0
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.15.0
-- Looking for ignition-fuel_tools4 -- found version 4.9.0
-- Searching for dependencies of ignition-fuel_tools4
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.5
[... 354 lines cut ...]
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/limits:827:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/type_traits:421:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/invoke.h:15:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/apply_cv.h:15:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/remove_reference.h:13:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstddef:46:5: error: <cstddef> tried including <stddef.h> but didn't find libc++'s <stddef.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
#   error <cstddef> tried including <stddef.h> but didn't find libc++'s <stddef.h> header. \
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstddef:59:9: error: no member named 'nullptr_t' in the global namespace
using ::nullptr_t;
      ~~^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:8:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/limits:827:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/type_traits:456:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_compound.h:14:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_fundamental.h:14:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_null_pointer.h:24:49: error: use of undeclared identifier 'nullptr_t'; did you mean 'nullptr'?
template <>          struct __is_nullptr_t_impl<nullptr_t> : public true_type {};
                                                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_null_pointer.h:24:49: error: template argument for template type parameter must be a type
template <>          struct __is_nullptr_t_impl<nullptr_t> : public true_type {};
                                                ^~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_null_pointer.h:23:17: note: template parameter is declared here
template <class _Tp> struct __is_nullptr_t_impl       : public false_type {};

ryanjAA, 2023-11-16:
continued:

```
                ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:8:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/limits:827:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/type_traits:538:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstdint:149:5: error: <cstdint> tried including <stdint.h> but didn't find libc++'s <stdint.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
#   error <cstdint> tried including <stdint.h> but didn't find libc++'s <stdint.h> header. \
    ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:9:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/string:545:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:28:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstring:66:5: error: <cstring> tried including <string.h> but didn't find libc++'s <string.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
#   error <cstring> tried including <string.h> but didn't find libc++'s <string.h> header. \
    ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:9:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/string:545:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:656:34: error: use of undeclared identifier 'nullptr_t'; did you mean 'nullptr'?
struct _LIBCPP_TEMPLATE_VIS hash<nullptr_t>
                                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:656:34: error: template argument for template type parameter must be a type
struct _LIBCPP_TEMPLATE_VIS hash<nullptr_t>
                                 ^~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:648:17: note: template parameter is declared here
template <class _Tp>
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:657:29: error: use of undeclared identifier 'nullptr_t'; did you mean 'nullptr'?
  : public __unary_function<nullptr_t, size_t>
                            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:657:29: error: template argument for template type parameter must be a type
  : public __unary_function<nullptr_t, size_t>
                            ^~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/unary_function.h:45:17: note: template parameter is declared here
template <class _Arg, class _Result>
                ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:9:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/string:551:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__iterator/reverse_iterator.h:18:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__iterator/advance.h:26:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstdlib:90:5: error: <cstdlib> tried including <stdlib.h> but didn't find libc++'s <stdlib.h> header.  [... cut ...]
FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build 
cd /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic && /usr/local/Cellar/cmake/3.27.8/bin/cmake --build /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -- -j 3
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

```

ryanjAA, 2023-11-16:
Then pointing to the version 12 sdk:

```
ryanjohnston@Ryans-MacBook-Pro PX4-Autopilot %HEADLESS=1 make px4_sitl gazebo-classic_plane_lidar
[0/954] git submodule src/drivers/gps/devices
[8/954] git submodule src/modules/mavlink/mavlink
[11/954] git submodule Tools/simulation/gazebo-classic/sitl_gazebo-classic
[12/954] git submodule src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client
[18/954] Performing configure step for 'sitl_gazebo-classic'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.58") found components: system thread filesystem 
-- Found DART: /usr/local/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Looking for ignition-math6 -- found version 6.15.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib
-- Looking for ignition-math6 -- found version 6.15.0
-- Looking for ignition-transport8 -- found version 8.4.0
-- Searching for dependencies of ignition-transport8
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.5
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.15.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Looking for ignition-common3 -- found version 3.16.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.16.0
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.15.0
-- Looking for ignition-fuel_tools4 -- found version 4.9.0
-- Searching for dependencies of ignition-fuel_tools4
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.5
-- Checking for module 'libzip'
--   Found libzip, version 1.10.1
-- Looking for ignition-common3 -- found version 3.16.0
-- Looking for ignition-math6 -- found version 6.15.0
-- Looking for ignition-msgs5 -- found version 5.11.0
CMake Warning (dev) at CMakeLists.txt:54 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

[... 326 lines cut ...]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cmath:549:46: note: candidate function
inline _LIBCPP_INLINE_VISIBILITY long double hypot( long double __x, long double __y, long double __z ) { return sqrt(__x*__x + __y*__y + __z*__z); }
                                             ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cmath:587:17: error: expected unqualified-id
    return std::isnan(__lcpp_x);
                ^
/usr/local/include/math.h:166:5: note: expanded from macro 'isnan'
[... cut ...]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstddef:46:5: error: <cstddef> tried including <stddef.h> but didn't find libc++'s <stddef.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
#   error <cstddef> tried including <stddef.h> but didn't find libc++'s <stddef.h> header. \

```

ryanjAA, 2023-11-16:
```

    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstddef:59:9: error: no member named 'nullptr_t' in the global namespace
using ::nullptr_t;
      ~~^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:8:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/limits:827:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/type_traits:456:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_compound.h:14:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_fundamental.h:14:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_null_pointer.h:24:49: error: use of undeclared identifier 'nullptr_t'; did you mean 'nullptr'?
template <>          struct __is_nullptr_t_impl<nullptr_t> : public true_type {};
                                                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_null_pointer.h:24:49: error: template argument for template type parameter must be a type
template <>          struct __is_nullptr_t_impl<nullptr_t> : public true_type {};
                                                ^~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__type_traits/is_null_pointer.h:23:17: note: template parameter is declared here
template <class _Tp> struct __is_nullptr_t_impl       : public false_type {};
                ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:8:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/limits:827:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/type_traits:538:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstdint:149:5: error: <cstdint> tried including <stdint.h> but didn't find libc++'s <stdint.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
#   error <cstdint> tried including <stdint.h> but didn't find libc++'s <stdint.h> header. \
    ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:9:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/string:545:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:28:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/cstring:66:5: error: <cstring> tried including <string.h> but didn't find libc++'s <string.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
#   error <cstring> tried including <string.h> but didn't find libc++'s <string.h> header. \
    ^
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.cc:4:
In file included from /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic/Odometry.pb.h:9:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/string:545:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:656:34: error: use of undeclared identifier 'nullptr_t'; did you mean 'nullptr'?
struct _LIBCPP_TEMPLATE_VIS hash<nullptr_t>
                                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:656:34: error: template argument for template type parameter must be a ty [... cut ...]
FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build 
cd /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic && /usr/local/Cellar/cmake/3.27.8/bin/cmake --build /Users/ryanjohnston/src/PX4NOVMASTER/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -- -j 3
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

ryanjAA, 2023-11-16:
@danielhonies

ryanjAA, 2023-11-16:
Some progress... First, when changing xcode-select to command line tools, the headers are found. So thats good. (sudo xcode-select -s /Library/Developer/CommandLineTools)

I also backed up the MacOS14.0.sdk into a .bak folder (go to /Library/Developer/CommandLineTools/SDKs) and mv MacOSX14.0sdk MacOSX14.0.sdk.bak (since 14 is a symlink) then added a symlink from SDK12.1 (sudo ln -s MacOSX12.1.sdk MacOSX14.0.sdk.

Still not building and lots of "error: no member named 'isless' in the global namespace" but we are finding headers... This is definitely a macOS issue but trying to find a way through...

mengchaoheng, 2024-03-01:
the same problem

mengchaoheng, 2024-04-16:
any suggestions? 
```Console
➜  PX4-Autopilot git:(main) make px4_sitl
-- PX4 version: v1.15.0-alpha1-775-g94d496605a (1.15.0)
CMake Warning (dev) at CMakeLists.txt:166 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.12.3", minimum required is "3")
-- PX4 config file: /Users/mch/Proj/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- The C compiler identification is AppleClang 15.0.0.15000309
-- The ASM compiler identification is AppleClang
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
CMake Deprecation Warning at platforms/posix/src/px4/common/lockstep_scheduler/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

/usr/local/opt/python@3.12/bin/python3.12: Error while finding module specification for 'symforce.symbolic' (ModuleNotFoundError: No module named 'symforce')
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2")
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.25
-- Found assimp: /usr/local/Cellar/assimp/5.4.0/include (found version "5.4.0")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found DART: /usr/local/include (Required is at least version "6.6") found components: dart
-- Found Boost: /usr/local/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libz.tbd (found version "1.2.12")
-- Looking for ignition-math6 -- found version 6.15.1
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
[... 533 lines cut ...]
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/__memory_resource/polymorphic_allocator.h:21:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/tuple:268:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/compare:145:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/__compare/compare_partial_order_fallback.h:13:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/__compare/partial_order.h:14:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/__compare/weak_order.h:14:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/__compare/strong_order.h:20:
/Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/include/c++/v1/cmath:321:5: error: <cmath> tried including <math.h> but didn't find libc++'s <math.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the c [... cut ...]
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build /Users/mch/Proj/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build 
cd /Users/mch/Proj/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic && /usr/local/Cellar/cmake/3.29.2/bin/cmake --build /Users/mch/Proj/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -- -j 2
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
➜  PX4-Autopilot git:(main) 
```

michaelgruner, 2024-04-19:
I ran into this problem elsewhere (nothing related to this project), and after hours of frustration I managed to build by switching the SDK to Xcode:

```bash
sudo xcode-select --switch /Applications/Xcode.app
```

Hope it helps.

mengchaoheng, 2024-04-20:
> I ran into this problem elsewhere (nothing related to this project), and after hours of frustration I managed to build by switching the SDK to Xcode:
> 
> ```shell
> sudo xcode-select --switch /Applications/Xcode.app
> ```
Thank you, this saved me !!!

hamishwillee, 2024-04-21:
@Jaeyoung-Lim If necessary, can we get an update to the PX4 mac OS installer for ^^^ ?

Jaeyoung-Lim, 2024-04-22:
@hamishwillee Would be nice, but the problem is that I don't have access to a Mac

mengchaoheng, 2024-04-22:
Some references https://github.com/PX4/PX4-Autopilot/issues/22826#issuecomment-2067587623

kadirozgun, 2024-04-24:
When I run on OSX 14.4.1 I had output:

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.10
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
[Wrn] [GuiIface.cc:114] Populating font family aliases took 84 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost. 
gzclient(25064,0x205699240) malloc: Incorrect checksum for freed object 0x7fbeb2874eb8: probably modified after being freed.
Corrupt value: 0x20a800000000
gzclient(25064,0x205699240) malloc: *** set a breakpoint in malloc_error_break to debug

Do you have any idea about this?

hamishwillee, 2024-04-24:
@mrpollo It looks like macOS Sonoma build needs some special massaging - the addition of a line to the installed dependencies as per  https://github.com/PX4/PX4-SITL_gazebo-classic/issues/1021#issuecomment-2067353768, and with some discussion linked in https://github.com/PX4/PX4-Autopilot/issues/22826#issuecomment-2067587623

But I don't even know where the mac OS dev toolchain scripts are hosted. Is this something you can fix, or address in the community call? 

PS I would like to know where the macOS dev toolchain scripts are hosted, but after 7 or 8 years of asking, never had an answer.

ryanjAA, 2024-04-24:
Part of me thinks the solution to macOS builds is some specific sdk versioning per px4 release tag (because you can run any version of macOS sdk you want easily via downloading an older one and symlinking to it) and using a venv…. That venv fixes 90% of toolchain issues once the sdk is sorted… that should ultimately be the path. 

@hamishwillee to preempt your question, I have to actually write this down since it’s what we always do but not document it but we always get it to build once doing those two things. macOS sdk and venv to override toolchain specifics in the environment.

hamishwillee, 2024-04-25:
@ryanjAA Its a nice idea, but how can we have such an SDK if we don't even know what the current installer does? - I repeat "I would like to know where the macOS dev toolchain scripts are hosted, but after 7 or 8 years of asking, never had an answer."

The best installers are turnkey "one click installer" solutions because they remove all the irrelevant setup crap and let people just start developing.  I'm happy with any solution that delivers that :-). 

Upshot, if you know how to get this working, then some documents would be appreciated. We're also getting users wondering when we'll support the new mac processors

ryanjAA, 2024-04-25:
The install script just installs dependencies but what it needs to do is install them in a virtual environment to not fight with local toolchain variables and a one line activation to activate. Im not sure you can create and activate a venv from a script but will try. 

As for the sdk we can match release to ask version then maybe include it in the script. 

Re where it is hosted, it’s just on the dev docs. No idea so controls or manages that. 

We know how to get it working so just need to document it. But for Mac we don’t use apple silicon for this exact purpose but iirc it does actually work.

ryanjAA, 2024-04-26:
Ya. What he said ⬆️⬆️⬆️

basically it’s a convoluted mess.

vjt, 2025-03-29:
> I ran into this problem elsewhere (nothing related to this project), and after hours of frustration I managed to build by switching the SDK to Xcode:
> 
> sudo xcode-select --switch /Applications/Xcode.app
> Hope it helps.

Thank you! This worked for me! 👍 👍 👍

hamishwillee, 2025-04-02:
@ryanjAA Is there a way to untangle this mess? An update to https://github.com/PX4/PX4-Autopilot/blob/main/Tools/setup/macos.sh , or an update to https://docs.px4.io/main/en/dev_setup/dev_env_mac.html ?

I can't drive this because I don't have a mac. But if we can fix up the tooling and have a PR on the docs, I can review it.

hamishwillee, 2025-04-02:
Yes, wouldn't it be nice if we had an always-on online simulator. As you say though, not really useful for average mac user picking this stuff up. 

Do we know anyone else who cares about macOS? Perhaps a group of you could collaborate.

ryanjAA, 2025-04-02:
Funny you say that.

We actually have an always online sim you can push code to but it spins up an AWS instance per user/everytime you push code (then kills the instance) so it's definitely not cost free and for sim work it would cost Dronecode quite a lot to no real benefit other than convenience... Definitely works though, you push code builds then uses AWS to serve and spits out a connection, etc etc.

Cares? lol. It's like fixed-wing aka the forgotten stepchild as TJ calls it... hahah 

Let me think about that one (or we just poll Discord for active Mac users).

hamishwillee, 2025-04-02:
>Let me think about that one (or we just poll Discord for active Mac users).

Thank you. I'll also ping within Auterion.

MeethiKishori, 2025-05-18:
@mrpollo , Have you already solved this issue in any recent build?