PX4/PX4-SITL_gazebo-classic #1032: About Lift Drag Plugin

gurselturkeri, 2024-03-04:
Hi, I am using PX4 Autopilot with Gazebo. I have edited vehicle param iris to advanced_plane which is in `mavros_posix_sitl.launch`. When I am launching file I can control fixed wing vehicle on QGrounControl but I am not sure simulation take into account lift drag plugin param such as CL0, CDa. How can I check that?
```
<plugin name="aircraft" filename="libAdvancedLiftDragPlugin.so">
      <a0>0.0</a0>
      <CL0>0.15188</CL0>
      <AR>6.5</AR>
      <eff>0.97</eff>
      <CLa>5.015</CLa>
      <CD0>0.029</CD0>
      <Cem0>0.075</Cem0>
      <Cema>-0.463966</Cema>
      <CYb>-0.258244</CYb>
      <Cellb>-0.039250</Cellb>
      <Cenb>0.100826</Cenb>
      <CDp>0.0</CDp>
      <CYp>0.065861</CYp>
      <CLp>0.0</CLp>
      <Cellp>-0.487407</Cellp>
      <Cemp>0.0</Cemp>
      <Cenp>-0.040416</Cenp>
      <CDq>0.055166</CDq>
      <CYq>0.0</CYq>
      <CLq>7.971792</CLq>
      <Cellq>0.0</Cellq>
      <Cemq>-12.140140</Cemq>
      <Cenq>0.0</Cenq>
      <CDr>0.0</CDr>
      <CYr>0.230299</CYr>
      <CLr>0.0</CLr>
      <Cellr>0.078165</Cellr>
      <Cemr>0.0</Cemr>
      <Cenr>-0.089947</Cenr>
      <alpha_stall>0.3391428111</alpha_stall>
      <CLa_stall>-3.85</CLa_stall>
      <CDa_stall>-0.9233984055</CDa_stall>
      <Cema_stall>0</Cema_stall>
      <ref_pt>-0.12 0.0 0.0</ref_pt>
      <area>0.34</area>
      <mac>0.22</mac>
      <air_density>1.2041</air_density>
      <forward>1 0 0</forward>
      <upward>0 0 1</upward>
      <link_name>base_link</link_name>
      <topic_name>lift_force/lumped_force</topic_name> <!--Uncomment to draw the force-->
      <num_ctrl_surfaces>4</num_ctrl_surfaces>
      <control_surface>
        <name>left_elevon_joint</name>
        <index>0</index>
        <direction>1</direction>
        <CD_ctrl>-0.000059</CD_ctrl>
        <CY_ctrl>0.000171</CY_ctrl>
        <CL_ctrl>-0.011940</CL_ctrl>
        <Cell_ctrl>-0.003331</Cell_ctrl>
        <Cem_ctrl>0.001498</Cem_ctrl>
        <Cen_ctrl>-0.000057</Cen_ctrl>
      </control_surface>
      <control_surface>
        <name>right_elevon_joint</name>
        <direction>1</direction>
        <index>1</index>
        <CD_ctrl>-0.000059</CD_ctrl>
        <CY_ctrl>-0.000171</CY_ctrl>
        <CL_ctrl>-0.011940</CL_ctrl>
        <Cell_ctrl>0.003331</Cell_ctrl>
        <Cem_ctrl>0.001498</Cem_ctrl>
        <Cen_ctrl>0.000057</Cen_ctrl>
      </control_surface>
      <control_surface>
        <name>elevator_joint</name>
        <direction>-1</direction>
        <index>2</index>
        <CD_ctrl>0.000274</CD_ctrl>
        <CY_ctrl>0</CY_ctrl>
        <CL_ctrl>0.010696</CL_ctrl>
        <Cell_ctrl>0.0</Cell_ctrl>
        <Cem_ctrl>-0.025798</Cem_ctrl>
        <Cen_ctrl>0.0</Cen_ctrl>
      </control_surface>
      <control_surface>
        <name>rudder_joint</name>
        <direction>1</direction>
        <index>3</index>
        <CD_ctrl>0.0</CD_ctrl>
        <CY_ctrl>-0.003913</CY_ctrl>
        <CL_ctrl>0.0</CL_ctrl>
        <Cell_ctrl>-0.000257</Cell_ctrl>
        <Cem_ctrl>0.0</Cem_ctrl>
        <Cen_ctrl>0.001613</Cen_ctrl>
      </control_surface>
      <robotNamespace></robotNamespace>
      <windSubTopic>world_wind</windSubTopic>
    </plugin>
```

Jaeyoung-Lim, 2024-03-04:
@gurselturkeri You can look at how the liftdrag forces are being computed here: https://github.com/PX4/PX4-SITL_gazebo-classic/blob/main/src/liftdrag_plugin/liftdrag_plugin.cpp

gurselturkeri, 2024-03-04:
> @gurselturkeri You can look at how the liftdrag forces are being computed here: https://github.com/PX4/PX4-SITL_gazebo-classic/blob/main/src/liftdrag_plugin/liftdrag_plugin.cpp
> 
> a tf tree is specific to your ROS setup, depending on how you are running px4 with ROS. Therefore it is likely the problem is with the specific setup you are using.
Firstly, thank you for reply!
So this cpp file is getting data from model sdf file? 
I use [this](https://docs.px4.io/main/en/simulation/ros_interface.html) document for ROS px4 setup. Is there any method to see tf tree suitable form and visualization on Rviz.

hamishwillee, 2024-03-06:
Might be helpful - docs on advanced plane here https://docs.px4.io/main/en/sim_gazebo_gz/vehicles.html#advanced-plane and lift drag plugin here https://docs.px4.io/main/en/sim_gazebo_gz/tools_avl_automation.html