PX4/PX4-SITL_gazebo-classic #1033: Increasing plane size in Gazebo

gurselturkeri, 2024-03-04:
Hi, I am using px4 autopilot with gazebo simulator. I would like to increasing advaned_plane size to more realistic which is in sitl_gazebo-classic but it has a lot of parameter depends on each other such as mass, inertia, motor, CLa,CL0,CD0,AR etc. How can I do that? Thanks.

https://github.com/PX4/PX4-SITL_gazebo-classic/blob/f754540e714642fea897445e69a675245bc6306a/models/advanced_plane/advanced_plane.sdf.jinja

NickatETH, 2024-03-08:
I have a similar problem:  
I am trying to model a drone of 20kg. 
I have all the parameters like total mass, total inertia, CLa, CL0, AR etc. and they are most likely correct as I used multiple methods to verify. 

However I struggle to find the correct mass and inertia for the control surfaces/rotor puller/imu. 
When using very small values (I assume it is done like this in the advanced_plane) or using the approximate values calculated from the hardware, the plane starts to spin very fast, mainly around the z-axis. 
Increasing the mass and inertias leads to more stable, but impossible to control results. 

So, how should the inertias of the control surfaces be set and what is their effect on the force calculation?

Jaeyoung-Lim, 2024-03-08:
@NickatETH @gurselturkeri In principle, given that the assumption the model structure and model parameter values are correct, the behavior of the vehicle should be the same.

>  For instance I will set mass 650kg then what will do other parameters?

You need to identify them from the real vehicle. 

Note that you need to get ALL the values correct for your target model, not just the subset of the parameter values. If the behavior is different to your real vehicle, either some of the parameter values are not correct OR the what the advancedliftdrag plugin is simulating is not the appropriate model for your target vehicle.

NickatETH, 2024-03-08:
Just so I understand correct: 
The mass/inertia in the beginning (line 7-18) is the whole plane without control surfaces and the prop. 

The other ones are just the moving parts on the drone ( for example the elevator in line 287-298)

Is that correct?

 #https://github.com/PX4/PX4-SITL_gazebo-classic/blob/main/models/advanced_plane/advanced_plane.sdf.jinja

Jaeyoung-Lim, 2024-03-08:
@NickatETH 

> The mass/inertia in the beginning (line 7-18) is the whole plane without control surfaces and the prop.
The other ones are just the moving parts on the plane ( for example the elevator in line 287-298)

Yes, although I doubt that there is any merit on trying to model the inertia of the moving parts of the plane. Matching the inertia to your designed CAD model would not result in a realistic simulation. Gazebo physics are not accurate enough to assess performance/behavior of the vehicle.

You can read through how you can utilize AVL to generate a reasonable model of the vehicle here: https://docs.px4.io/main/en/sim_gazebo_gz/tools_avl_automation.html

> I addition, does the positioning of the control surfaces matter or is it just for visualisation?

The position itself is just visualization, but the way the plugin works is that it reads the joint values to calculate the forces induced by the control surface: https://github.com/PX4/PX4-SITL_gazebo-classic/blob/f754540e714642fea897445e69a675245bc6306a/src/liftdrag_plugin/advanced_liftdrag_plugin.cpp#L437