PX4/PX4-SITL_gazebo-classic #1053: No BARO and no MAG from r1_rover in gazebo

matiasguzmanp, 2024-08-20:
I'm trying to launch the simulation of the r1_rover out-of-the-box following this [page](https://docs.px4.io/main/en/sim_gazebo_classic/). 

In first instance I get this error

ERROR [simulator_mavlink] poll timeout 0, 22
poll timeout

[image]

I found that to solve this issue (or at least to circumvent it) we can change the parameter of lockstep in the sdf file of r1_rover and in the config of PX4. Once I did this i got another error.

ERROR [vehicle_air_data] BARO #0 failed:  TIMEOUT!	
ERROR [vehicle_magnetometer] MAG #0 failed:  TIMEOUT!

[image]

I didn't found a solution to this error, can I get any help ?

Ozkahraman, 2025-08-04:
Hello Matias, I also encountered this error. When I removed the lockstep line from the .jinja file, the error disappeared, but the Gazebo simulation started lagging—so it wasn’t really a solution. I just wanted to ask if you were able to solve this issue.

matiasguzmanp, 2025-08-05:
Hi Ozkahraman! 

Unfortunately, I wasn’t able to find a solution to the problem. Since my project was under significant time constraints, I had to switch to Ardupilot instead. Sorry.