PX4/PX4-SITL_gazebo-classic #1059: No geotagged_cam image saved?

tnedev, 2024-11-25:
Hey guys, 

According to https://docs.px4.io/main/en/sim_gazebo_classic/#simulated-survey-camera, during camera trigger, a geotagged image should be saved in the build files. 

However, I don't see a saved image anywhere.  I'm using all defaults with plane_cam model. I see the MAVLlink trigger message, but no files. 

Is there something that I'm missing?

I'm running PX4 1.15.2 and Gazebo Classic under docker on Ubuntu 20.

hamishwillee, 2024-11-25:
I don't think you're missing anything documented, so if you're sure that you've followed the instructions properly I would ask on the discussion boards and do a search on PX4-Autopilot bug list. Note that this might not get fixed as Gazebo-Classic is approaching end of life, but I'd expect something similar to work with Gazebo.

I'll leave this open - might find time to test it in the next couple of days.

tnedev, 2024-11-26:
Thank you @hamishwillee. 
I noticed that taking an image with the image capture button on QGC, actually saves the file, but while doing survey it does not work. So it might be a MAVLink issue.

hamishwillee, 2024-11-27:
Its never a MAVLink issue :-). Only usage of MAVLink!

hamishwillee, 2024-11-27:
OK, so I had a quick play on main. The camera_image_captured message is being received by QGC which means that the camera trigger module knows it is getting a capture message. I'm not seeing the images being created either though.

@Jaeyoung-Lim Is image capture in Gazebo classic simulation something you know about?

tnedev, 2024-11-27:
@hamishwillee Thanks for the help. 
BTW, when the manual image capture button is pressed, image are written to _build/px4_sitl_default/src/modules/simulation/simulator_mavlink/frames/_

NOT to: _build/px4_sitl_default/tmp/frames/_ as written in the documentation.

hamishwillee, 2024-11-27:
Docs are probably in error, but lets see if we @Jaeyoung-Lim can confirm.

hamishwillee, 2024-12-04:
@tnedev I hear from @Jaeyoung-Lim this is expected to work. So can you raise an issue in PX4-Autopilot and cross link to here. It doesn't work for me either.

Jaeyoung-Lim, 2024-12-06:
Fixed in https://github.com/PX4/PX4-Autopilot/pull/24081

closed 2024-12-06 by Jaeyoung-Lim

hamishwillee, 2024-12-08:
Thanks so much for sorting this @Jaeyoung-Lim !

hamishwillee, 2024-12-12:
FYI docs also fixed for location where frames are captured in https://github.com/PX4/PX4-user_guide/pull/3493