PX4/PX4-SITL_gazebo-classic #1060: Mavlink plugin not working in custom urdf

crazy-robotics, 2025-01-13:
I have custom urdf file which I make by following [discussion](https://discuss.px4.io/t/create-custom-model-for-sitl/6700) to make my custom sitl vehicle work with Gazebo simulation and QGC so I can test my offboard code. I built my custom model by `make px4_sitl gazebo-classic_my_custom_uav` the drone launch in the gazebo but not connected to QGC and mavlink[info] messages are not showing on the terminal 
```
<?xml version="1.0" encoding="utf-8"?>
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.6.0-4-g7f85cfe  Build Version: 1.6.7995.38578
     For more information, please see http://wiki.ros.org/sw_urdf_exporter -->
<robot
  name="my_custom_uav">
  <link
    name="base_link">
    <inertial>
      <origin
        xyz="0.0066871 0.00057296 0.071314"
        rpy="0 0 0" />
      <mass
        value="2.2494" />
      <inertia
        ixx="0.024649"
        ixy="-7.8701E-06"
        ixz="-4.0429E-06"
        iyy="0.024447"
        iyz="-1.0355E-05"
        izz="0.046713" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://my_custom_uav/meshes/base_link.STL" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://my_custom_uav/meshes/base_link.STL" />
      </geometry>
    </collision>
  </link>
  <link
    name="m1">
    <inertial>
      <origin
        xyz="-0.000275758863509959 -0.00027577321010086 -0.0288987241999298"
        rpy="0 0 0" />
      <mass
        value="0.0102857652690979" />
      <inertia
        ixx="1.4471352439084E-06"
        ixy="2.46724635603962E-11"
        ixz="2.61968213356202E-09"
        iyy="1.44713594954588E-06"
        iyz="2.61993099804036E-09"
        izz="1.72167745600157E-06" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
[... 722 lines cut ...]
          <joint_control_pid>
            <p>10.0</p>
            <i>0</i>
            <d>0</d>
            <iMax>0</iMax>
            <iMin>0</iMin>
            <cmdMax>20</cmdMax>
            <cmdMin>-20</cmdMin>
          </joint_control_pid>
        </channel>
        <channel name='m8'>
          <input_index>7</input_index>
          <input_offset>0</input_offset>
          <input_scaling>0.524</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position</joint_control_type>
        </channel>
      </control_channels>
    </plugin>
    <!-- <static>0</static> -->

</robot>
```
[image]

hamishwillee, 2025-01-19:
@Jaeyoung-Lim You're the right person to look at issues in this repo right?

Jaeyoung-Lim, 2025-01-20:
@hamishwillee Yes,

@crazy-robotics Unfortunately it is hard for me to debug custom implementations. Please ask specific questions related to this repository that would be useful for debugging your problem.

crazy-robotics, 2025-01-20:
Thanks, seniors for replying I fixed it by manually looking at my URDF file generated from [solidworks to urdf exporter](https://wiki.ros.org/sw_urdf_exporter) and replacing in iris model now I facing the issue drone arm by QGC and do not taking off I don't know may be due to over weight or what look at the [issue](https://discuss.px4.io/t/drone-arm-but-not-takeoff-sitl-gazebo/43349). masses and material of the drone carefully selected in Solidworks replica of original drone