PX4/PX4-SITL_gazebo-classic #1063: In multi-machine simulation, the gimbal attitude information of several typhoon_h480 is mixed in one message.

Yingqiao1998, 2025-02-25:
Hi, I tried to control Typhoon's gimbal and read its attitude in the multi-machine simulation:

Tools/simulation/gazebo-classic/sitl_multiple_run.sh -s "typhoon_h480:3"

[image]

After read some issues, I changed some of the files accordingly:
Tools/simulation/gazebo-classic/sitl_multiple_run.sh：

[image]

Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/typhoon_h480/typhoon_h480.sdf.jinja：

[image]

Tools/simulation/gazebo-classic/sitl_gazebo-classic/scripts/jinja_gen.py：

[image]

then I can control different gimbal's attitude.
However, I cannot read there's attitude.

In QGC:

[image]

the Hz of this component means 10* vehicle number.
And no attitude message in plane system.

[image]

I read some simple of mavsdk, and I tried like this:

https://github.com/mavlink/MAVSDK/blob/main/examples/gimbal_full_control/gimbal_full_control.cpp

I can control it, but I still can't read the attitude.

I think there may some sysid problem in model starting.  Who can give me some advice? It's urgent.

SuFangzheng, 2025-02-26:
I have same problem!

Yingqiao1998, 2025-03-03:
I soloved this problem. Just because the lib "libgazebo_gimbal_controller_plugin" use "private: static constexpr uint8_t oursysid" as system id. So I changed some files for reading sysid from sdf:
in Tools/simulation/gazebo-classic/sitl_gazebo-classic/include/gazebo_gimbal_controller_plugin.hh:

[image]

in Tools/simulation/gazebo-classic/sitl_gazebo-classic/src/gazebo_gimbal_controller_plugin.cpp:

[image]

in Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/typhoon_h480/typhoon_h480.sdf.jinja:

[image]

in Tools/simulation/gazebo-classic/sitl_gazebo-classic/scripts/jinja_gen.py：
add 2 line of gimbal_sysid

[image]

in Tools/simulation/gazebo-classic/sitl_multiple_run.sh:

[image]

closed 2025-03-03 by Yingqiao1998